commaai/opendbc #21: Cabana strips CHFFR_METRIC comment

energee, 2017-09-17:
Since this directly relates to DBC files, this seems like the right place.

When saving to github from cabana: CM_ "CHFFR_METRIC" gets taken out of the file

energee, 2017-09-20:
I'm not seeing this anymore so either i was wrong or this is fixed.

energee, 2017-09-20:
Ok on looking at this a little further, there is some odd behaviour. The chffr_metric comment is not being stripped but this is happening:

The format is:

`CM_ "CHFFR_METRIC <message id> <signal> <signal> <value> <value>; CHFFR_METRIC <message id> <signal> <signal> <value> <value>";`

When saved, semicolon is stripped so the line above becomes:

`CM_ "CHFFR_METRIC <message id> <signal> <signal> <value> <value>; CHFFR_METRIC <message id> <signal> <signal> <value> <value>"`

On subsequent saves, quotes get added to the line, so on next save it would look like:

`CM_ "CHFFR_METRIC <message id> <signal> <signal> <value> <value>; CHFFR_METRIC <message id> <signal> <signal> <value> <value>""`

andyh2, 2017-09-30:
Thanks for the report - should be fixed as of this evening in cabana. Let me know if you have any further issues saving chffr_metrics.

closed 2017-10-11 by energee