commaai/opendbc #95: CT6 LKAS Mode Signal

wangdino, 2018-05-19:
Hi,

I had a look at the **LKASMode** signal in this message:

```
BO_ 338 ASCMLKASteeringCmd: 6 NEO
 SG_ LKASteeringCmdActive : 7|1@0+ (1,0) [0|0] ""  NEO
 SG_ LKASteeringCmd : 5|14@0- (1,0) [0|0] ""  NEO
 SG_ RollingCounter : 23|2@0+ (1,0) [0|0] ""  NEO
 SG_ SetMe1 : 21|1@0+ (1,0) [0|0] ""  NEO
 SG_ LKASVehicleSpeed : 20|13@0+ (0.22,0) [0|0] "kph" NEO
 SG_ LKASMode : 36|2@0+ (1,0) [0|0] ""  NEO
 SG_ LKASteeringCmdChecksum : 33|10@0+ (1,0) [0|0] ""  NEO
```

It seems value=2 represents SuperCruise on. Any idea what value=1 means? 

Thanks for any insights!

rbiasini, 2018-05-19:
VAL 1 is standard lkas. See VAL definitions at the end of the file.

closed 2018-05-19 by rbiasini

wangdino, 2018-05-19:
I asked because most of the time the signal was 0 when SuperCruise wasn't turned on. It only flashes to 1 a few times in a fraction of a second...

rbiasini, 2018-05-19:
yeah, it's because standard LKAS intervenes only when going out of the lane. Interventions are very sporadic