commaai/opendbc #150: Reverse engineering question

gianpyc, 2019-03-27:
Very interesting project.
However, how can I create a dbc like these available in this repo? Is Panda that generates the dbc according to the frames sniffed or another tool?

Thanks

rbiasini, 2019-03-27:
There is no automatic tool to autogenerate dbc files. You need to reverse engineer the meaning of each message/signal. Panda is just a tool to be able to read the CAN traffic from a vehicle.

gianpyc, 2019-03-27:
Got it.
Can I see an example of what Pange sniffs? Just to understand how then to generate the corresponding signal in the dbc.

rbiasini, 2019-03-27:
you should join discord.comma.ai, lots of developers there who can help out.

dposea, 2019-04-18:
Guys, I applaud your efforts but there are other ways.  II used to own a major US tuning company, and I know just a tiny bit about Ford and GM communications. Ford used something called DPIDS, which were actual memory address in the ecu. I'm sure they still do that, at lo and high speed. On my 1996 Mustang 10 bytes of data were sent every ~8 ms in "rapid packet mode. You  had to set up the data before starting rhe process, and a max of I think 8 sets of 5 bytes could be defined. If there were multiple packets defined the first 10 bytes were sent, then the nexr 10 8 ms later, and so on until it would repeat until you sent a stop command. We needed address for every parameter on every vehicle Ford made. There are sources for this information, usually pretty expensive but a group could rasie funds if necessary. Or so I heard once.  I put enough detail into the description above so you know I'm not full of bs. I wrote datalogging software and firmware, and did a lot of reverse engineering.

dposea, 2019-04-18:
Can you contact me through my login here? I don't want to post an email too publically. I may be able to offer more advice on how to jumpstart an open database but I do need to be discreet.

closed 2020-05-28 by adeebshihadeh