commaai/opendbc #151: DBC files don't follow the spec

msalau, 2019-03-28:
Hello,

I tried to open the DBC files from the repo with CANdb++ Editor (the original tool used to create such files) and it fails.
The reason for this is that the files (or the generator) don't follow the spec.
Most of parsers found on Internet don't enforce order of statements in DBC files, but the original editor does.

The spec is not disclosed officially, so most of information is gathered through reverse engineering.

I know of 2 parsers that do enforce ordering of sections like the original editor does:

1. https://bitbucket.org/tobylorenz/vector_dbc (reads data into C++ structures)
2. https://github.com/msalau/dbc-parser (just prints scanned info to stdout)

The second one is mine, so I can help with it.

The original DBC parser expects the following ordering (all the sectional are optional if not stated otherwise):

1. VERSION
2. NS_ : ...
3. BS_: (mandatory)
4. BU_: (mandatory)
5. VAL_TABLE_
6. BO_ and SG_
7. CM_
8. BA_DEF_
9. BA_DEF_DEF_
10. BA_
11. VAL_

Sections may not mix.
There are few more sections not listed here, but their use is less widespread.

Hope that helps.

Regards,
Maksim

msalau, 2019-03-31:
BTW,

As I understood, some comments are used to preprocess files (e.g. `CM_ "IMPORT _honda_2017.dbc"`). There is another mechanism to put statements into DBC files that will be ignored.
Text appearing after `//` up to the end on line will be ignored by CANdb++. Just like C/C++ style comments.
E.g.:
`// IMPORT _honda_2017.dbc`

Regards,
Maksim

P.S. Also please note that `CM_ "IMPORT _honda_2017.dbc"` is not a valid DBC statement disregarding place where it appears. `CM_` statements must be terminated with `;`

rbiasini, 2019-03-31:
@msalau , thanks, I agree that dbc files should be changed to follow a standard. It's not high priority for us to fix it now, but fell free to open a PR to address some/all the inconsistencies you see and we'll merge it in.

msalau, 2019-04-04:
Hi @rbiasini 

It turned out there is already a PR to fix the issues in generated files: #113 

As for the spec, the following document may be found: [link](http://read.pudn.com/downloads766/ebook/3041455/DBC_File_Format_Documentation.pdf)
It is outdated and incomplete, but covers basic functionality and requirements.

Regards,
Maksim

Clubsport, 2019-12-23:
I also get error messages when I try to open vw_mqb_2010.dbc , may someone would like to fix this as a Christmas gift? :o)