commaai/opendbc #197: How can I convert can log(.csv) to can signal???

DonghunP, 2019-11-08:
I want to convert can log file(.csv) to can signal.

[link](https://my.comma.ai/cabana/?route=aa290427b48a6599%7C2019-11-07--06-58-58&exp=1604656007&sig=bqaOMffWKZFke7rbUAOZKPen8dgAo%2FLBKzcCFHZnjy4%3D&max=31&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Faa290427b48a6599%2F28b5e3985da92f223eeecb6af53ab53e_2019-11-07--06-58-58)  
In this link, can log file can be converted to can message.

How can I do it??

Thanks ahead.

adeebshihadeh, 2020-05-11:
A can log is just the raw data. You need to reverse engineer the signals yourself and define them in a DBC file.

closed 2020-05-11 by adeebshihadeh

adhintz, 2020-05-11:
If you want to replay the saved CSV file so that the Panda sends the mesages, here's a script to do that: https://github.com/adhintz/panda/blob/949b1b29b6d45de1183898057ca643fd914946c3/examples/can_replay.py