commaai/opendbc #200: Question on hyundai_kia_generic.dbc

gianpyc, 2019-11-25:
Hi,
I'm analyzing the hyundai_kia_generic DBC and I've a question on the meaning of this BO_
TMU_GW_PE_01

My guess is that the TMU is the telematic ECU, but GW? Is it referred to the gateway? And PE?

Thanks

bbsusheelkumar, 2019-11-25:
In general, 
TMU is Telematics ECU
GW stands for Gateway
PE stands for Periodic and Event based.

This generally means message is getting transmitted from TMU to GW and its transmission is both periodic and event based.

gianpyc, 2019-11-25:
Ok, but since the signals are received by the DATC, this means that the frame is sent from the TMU to the DATC passing through the GW?

In HU_DATC_PE_00, HU stands for? In general where I can find the list of the acronyms?

bbsusheelkumar, 2019-11-25:
In general HU stands for Head Unit i.e.the infotainment of car.

gianpyc, 2019-11-26:
Ok,
but what is not clear to me in TMU_GW_PE_01 is that you said that the message is sent from TMU to GW, instead in the dbc says that signals are sent from CLU to DATC, where am I wrong? Tnx

bbsusheelkumar, 2019-11-27:
CLU and DATC in the DBC refers to Nodes. TMU and HU related messages are generally part of CLU node. And it is sent to DATC node through Gateway.

closed 2020-05-11 by adeebshihadeh