commaai/opendbc #303: All Mercedes vehicles between 2000-2010

rnd-ash, 2020-10-08:
Hi, As part of a project, I've managed to reverse engineer binary files found in Xentry to obtain all this CAN data about all MB vehicles between 2000-2010. The 2 files attached are for the CAN B and C network for my MB model (W203)

I am too lazy to convert them to dbc...so just putting them here if anyone wants to convert them

[203_b.txt](https://github.com/commaai/opendbc/files/5347494/203_b.txt)
[203_c.txt](https://github.com/commaai/opendbc/files/5347495/203_c.txt)

You can find the extraction script here:
https://github.com/rnd-ash/MBUX-Port/blob/master/parse_dat_v3.py

rnd-ash, 2020-10-08:
Unfortuantly, for some values that dont map directly for 1 to 1, there is no way for the extractor script to convert from a raw int to something like Temp C...Below are a couple examples showing that the values don't map directly 1 to 1:
Interior temp is (Raw) / 2 - 40
Fuel usage is (RAW)/2
Exterior temp is Raw-40

mitchdetailed, 2021-05-12:
I may take this on. :) been a bit lazy with my e55 amg lately

adeebshihadeh, 2021-09-02:
Feel free to make a PR for this, but I'll close since it's not an issue.

closed 2021-09-02 by adeebshihadeh