commaai/opendbc #416: generator: merge partial message definitions

belm0, 2021-08-22:
I'd like the generator to be able to merge multiple partial message definitions into a coherent output.  (Currently, I believe a redefined message will silently overwrite the entire previous definition.)  It should raise an error if the merge inputs have duplicate or overlapping fields.

I wanted to use this when refactoring honda files to share common definitions for Fit platform models as much as possible.

wanted
-----
in `_honda_fit_2019.dbc`:
```
BO_ 422 SCM_BUTTONS: 8 SCM
 SG_ CRUISE_BUTTONS : 7|3@0+ (1,0) [0|7] "" EON
 SG_ LIGHTS_SETTING : 1|2@0+ (1,0) [0|3] "" EON
 SG_ MAIN_ON : 47|1@0+ (1,0) [0|1] "" EON
 SG_ CRUISE_SETTING : 43|2@0+ (1,0) [0|3] "" EON
 SG_ COUNTER : 61|2@0+ (1,0) [0|3] "" EON
 SG_ CHECKSUM : 59|4@0+ (1,0) [0|3] "" EON
```

in `honda_hrv_touring_2019_can.dbc`
```
CM_ "IMPORT _honda_fit_2018.dbc";

BO_ 422 SCM_BUTTONS: 8 SCM
 SG_ DRIVERS_DOOR_OPEN : 63|1@0+ (1,0) [0|1] "" EON
```

current situation
----------
I cannot put the `SCM_BUTTONS` message definition in the common file.  Since some models have an added field, the entire message must be defined in each top-level dbc.

belm0, 2021-08-23:
question: is it harmless to define message fields that are only available on some car models (DRIVERS_DOOR_OPEN above)?

if that's allowed, then I can see why there hasn't been demand for a merging feature, because we can just define the complete message in a common file

```
BO_ 422 SCM_BUTTONS: 8 SCM
 SG_ CRUISE_BUTTONS : 7|3@0+ (1,0) [0|7] "" EON
 SG_ LIGHTS_SETTING : 1|2@0+ (1,0) [0|3] "" EON
 SG_ MAIN_ON : 47|1@0+ (1,0) [0|1] "" EON
 SG_ CRUISE_SETTING : 43|2@0+ (1,0) [0|3] "" EON
 SG_ DRIVERS_DOOR_OPEN : 63|1@0+ (1,0) [0|1] "" EON
 SG_ COUNTER : 61|2@0+ (1,0) [0|3] "" EON
 SG_ CHECKSUM : 59|4@0+ (1,0) [0|3] "" EON
```

adeebshihadeh, 2021-09-02:
Yes, defining message fields only available on some models is allowed. Merging at the signal level is too complicated and will get messy pretty quickly.

closed 2021-09-02 by adeebshihadeh