commaai/opendbc #470: Errors/Inconsistencies things on few DBC files

maxcleme, 2021-11-18:
Hi,

I just wanted to report what I think are some errors/inconsistencies/invalid things on few DBC here.

While writing my own DBC parser, I end up with something working on almost everything here (I was using this repo as my testing oracle to see if everything was working fine on my end), however there is still few things I'm not able to parse, but I'm confident that my parser is correct.

I'll try to list in a diff format what I've done in order to make all files pass.

```diff
diff --git a/ford_lincoln_base_pt.dbc b/ford_lincoln_base_pt.dbc
-VAL_ 934 SodSnsLeft_D_Stat 3 "Second_Warning_Audio" 2 "System_Failure" 1 
+VAL_ 934 SodSnsLeft_D_Stat 3 "Second_Warning_Audio" 2 "System_Failure" 1 "" ;
```

> Misses a value description for value 1, also misses `;`.

<br>
<br>

```diff
diff --git a/ford_lincoln_base_pt.dbc b/ford_lincoln_base_pt.dbc
-CM_ VAL_ 970 Lkas_Alert 15 "no_alert" 3 "high_intensity" 2 "mid_intensity" 1 "low_intensity" ;
+VAL_ 970 Lkas_Alert 15 "no_alert" 3 "high_intensity" 2 "mid_intensity" 1 "low_intensity" ;
```

> Dunno why there is a `CM_` here, it makes no sense to be honest. Since it is in the middle of all `VAL_` declarations, I ended up deleting the suspicious `CM_` keyword. Moreover, value table are not eligible for comment.

<br>
<br>

```diff
diff --git a/ford_lincoln_base_pt.dbc b/ford_lincoln_base_pt.dbc
-VAL_ 972 LaActAvail_D_Actl 3 "LKA_LCA_LDW_Avail" 2 "LCA_LKA_Avail_LDW_Suppress" 1 "LCA_LKA_Suppress_LDW_Avail" 0
+VAL_ 972 LaActAvail_D_Actl 3 "LKA_LCA_LDW_Avail" 2 "LCA_LKA_Avail_LDW_Suppress" 1 "LCA_LKA_Suppress_LDW_Avail" 0 "" ;
```

> Misses a value description for value 0, also misses `;`.

<br>
<br>

```diff
diff --git a/ford_lincoln_base_pt.dbc b/ford_lincoln_base_pt.dbc
-VAL_ 984 AhbHiBeam_D_Rq 3 "NotUsed" 2 "HighBeamRecommended" 1 "LowBeamRecommended" 0 "DeactivatedUnavailable" ; "LCA_LKA_LDW_Suppress" ;
+VAL_ 984 AhbHiBeam_D_Rq 3 "NotUsed" 2 "HighBeamRecommended" 1 "LowBeamRecommended" 0 "DeactivatedUnavailable" ;
```

> There is a value description after the end of the statement. Anyway if it is needed, it also misses a value for it.

<br>
<br>

```diff
diff --git a/honda_civic_touring_2016_can_generated.dbc b/honda_civic_touring_2016_can_generated.dbc
-VAL_ 884 DASHBOARD_ALERT 0 "none" 51 "acc_problem" 55 "cmbs_problem" 75 "key_not_detected" 79 "fasten_seatbelt" 111 "lkas_problem" 131 "brake_system_problem" 132 "brake_hold_problem" 139 "tbd" 161 "door_open"
+VAL_ 884 DASHBOARD_ALERT 0 "none" 51 "acc_problem" 55 "cmbs_problem" 75 "key_not_detected" 79 "fasten_seatbelt" 111 "lkas_problem" 131 "brake_system_problem" 132 "brake_hold_problem" 139 "tbd" 161 "door_open" ;
```

> Misses `;`.

<br>
<br>
 
```diff
diff --git a/honda_clarity_hybrid_2018_can_generated.dbc b/honda_clarity_hybrid_2018_can_generated.dbc
-VAL_ 884 DASHBOARD_ALERT 0 "none" 51 "acc_problem" 55 "cmbs_problem" 75 "key_not_detected" 79 "fasten_seatbelt" 111 "lkas_problem" 131 "brake_system_problem" 132 "brake_hold_problem" 139 "tbd" 161 "door_open"
+VAL_ 884 DASHBOARD_ALERT 0 "none" 51 "acc_problem" 55 "cmbs_problem" 75 "key_not_detected" 79 "fasten_seatbelt" 111 "lkas_problem" 131 "brake_system_problem" 132 "brake_hold_problem" 139 "tbd" 161 "door_open" ;
```

> Misses `;`.

<br>
<br>

```diff
diff --git a/mazda_2017.dbc b/mazda_2017.dbc
-BO_ 1275 2017_5: 8 XXX
- SG_ counter : 4|5@0+ (1,0) [0|255] "" XXX
-
-BO_ 1070 2017_1: 8 XXX
-
-BO_ 1183 2017_2: 8 XXX
-
-BO_ 1243 2017_3: 8 XXX
- SG_ NEW_SIGNAL_1 : 7|64@0+ (1,0) [0|18446744073709552000] "" XXX
 
-BO_ 1178 2017_6: 8 XXX
- SG_ NEW_SIGNAL_1 : 7|64@0+ (1,0) [0|18446744073709552000] "" XXX
-
-BO_ 1179 2017_7: 8 XXX
- SG_ NEW_SIGNAL_1 : 7|64@0+ (1,0) [0|18446744073709552000] "" XXX
-
-BO_ 1435 2017_8: 8 XXX
-
```

> Those can't be valid signal name (`2017_X`).
> It has to starts with either `alpha_char` or `_`.

<br>
<br>

```diff
diff --git a/toyota_sienna_xle_2018_pt_generated.dbc b/toyota_sienna_xle_2018_pt_generated.dbc
-VAL_ 610 LKA_STATE 25 "temporary_fault" 9 "temporary_fault2" 5 "active" 1 "standby"
+VAL_ 610 LKA_STATE 25 "temporary_fault" 9 "temporary_fault2" 5 "active" 1 "standby";
```

> Misses `;`.

<br>
<br>

```diff
diff --git a/vw_mqb_2010.dbc b/vw_mqb_2010.dbc
-BU_: Airbag_MQB BAP_Tester_MQB BMS_MQB Datenlogger_MQB Gateway_MQB Getriebe_DQ_Hybrid_MQB Getriebe_DQ_MQB LEH_MQB Motor_Diesel_MQB Motor_Hybrid_MQB Motor_Otto_MQB SAK_MQB Waehlhebel_MQB Vector__XXX 9 l c i XXX
+BU_: Airbag_MQB BAP_Tester_MQB BMS_MQB Datenlogger_MQB Gateway_MQB Getriebe_DQ_Hybrid_MQB Getriebe_DQ_MQB LEH_MQB Motor_Diesel_MQB Motor_Hybrid_MQB Motor_Otto_MQB SAK_MQB Waehlhebel_MQB Vector__XXX l c i XXX
```

> According to specification, `9` can't be valid network node name.
> It has to starts with either `alpha_char` or `_`.

```diff
diff --git a/vw_mqb_2010.dbc b/vw_mqb_2010.dbc
-CM_ SG_ 159 EPS_HCA_Status "Status of Heading Control Assist feature"
+CM_ SG_ 159 EPS_HCA_Status "Status of Heading Control Assist feature";
```

> Misses `;`.

<br>
<br>

```diff
diff --git a/vw_mqb_2010.dbc b/vw_mqb_2010.dbc
-CM_ SG_ 159 EPS_VZ_BLW "Raw steering angle, direction"
+CM_ SG_ 159 EPS_VZ_BLW "Raw steering angle, direction";
```

> Misses `;`.

<br>
<br>

```diff
diff --git a/vw_mqb_2010.dbc b/vw_mqb_2010.dbc
-CM_ SG_ 294 3 "May be zero when sent by older cameras";
-CM_ SG_ 294 7 "May be zero when sent by older cameras";
-CM_ SG_ 294 254 "May be zero when sent by older cameras";
```

> According to specification, those can't be valid signal name (`3`, `7`, `254`).
> It has to starts with either `alpha_char` or `_`.

jyoung8607, 2021-11-18:
Thank you for the feedback. I'm sort of de facto maintainer of the VW DBC and port so I've filed a PR for those parts.

> ```diff
> diff --git a/vw_mqb_2010.dbc b/vw_mqb_2010.dbc
> -BU_: Airbag_MQB BAP_Tester_MQB BMS_MQB Datenlogger_MQB Gateway_MQB Getriebe_DQ_Hybrid_MQB Getriebe_DQ_MQB LEH_MQB Motor_Diesel_MQB [...]

Removed. That said, the network-node matrix is really not something I've maintained because openpilot doesn't use it, so I don't expect the rest of it to be all that useful.

> ```diff
> diff --git a/vw_mqb_2010.dbc b/vw_mqb_2010.dbc
> -CM_ SG_ 159 EPS_HCA_Status "Status of Heading Control Assist feature"
> +CM_ SG_ 159 EPS_HCA_Status "Status of Heading Control Assist feature";
> ```
> 
> > Misses `;`.

Fixed.

> ```diff
> diff --git a/vw_mqb_2010.dbc b/vw_mqb_2010.dbc
> -CM_ SG_ 159 EPS_VZ_BLW "Raw steering angle, direction"
> +CM_ SG_ 159 EPS_VZ_BLW "Raw steering angle, direction";
> ```
> 
> > Misses `;`.

Fixed.

> ```diff
> diff --git a/vw_mqb_2010.dbc b/vw_mqb_2010.dbc
> -CM_ SG_ 294 3 "May be zero when sent by older cameras";
> -CM_ SG_ 294 7 "May be zero when sent by older cameras";
> -CM_ SG_ 294 254 "May be zero [...]

They were also dangling references, the signal names were changed long ago. Fixed by changing the names to the ones currently used in HCA_01. As a separate matter, the whole HCA_01 message and signal structure needs to be overhauled. Doing so is on my long-term to-do list, but it requires touching a lot of things in a lot of places which won't happen today.

maxcleme, 2021-11-19:
@jyoung8607 

Glad it could help some people here, do you have any way to ping other maintainers ?

adeebshihadeh, 2021-11-19:
Generally, we close issues like this that should be PRs, especially when they don't affect openpilot functionality.

adeebshihadeh, 2021-12-21:
Closing since these don't affect openpilot functionality, but feel free to submit a PR to address these issues.

closed 2021-12-21 by adeebshihadeh