commaai/opendbc #496: Honda DBC cleanup

adeebshihadeh, 2022-01-18:
Ideally, we can use a single DBC for all Hondas, like we do with many of the more recent car ports. This can be done incrementally by doing the following:
* merging DBCs
* using common, universal signals when possible
* abstracting out repeated definitions into common DBCs

sshane, 2022-01-28:
- [ ] See if `LAT_ACCEL` under `KINEMATICS` on 145 is also available on 148 for cars using 145. The LAT_ACCEL signal is defined in the same location in the message
- [ ] Fix duplicate gear signals defined
- [ ] A lot of these messages are common in bosch vehicles, possible unify:
  - [ ] DOORS_STATUS
  - [ ] STANDSTILL
  - [ ] GEARBOX
  - [ ] GAS_PEDAL_2

sshane, 2022-01-28:
Notes about common signals:

- `DOORS_STATUS`:
  - Identical: `honda_civic_hatchback_ex_2017_can`, `honda_civic_sedan_16_diesel_2019_can`, `honda_crv_ex_2017_can`, `honda_insight_ex_2019_can`
  - Missing (using `SCM_FEEDBACK`): `honda_crv_hybrid_2019_can`, `honda_accord_2018_can`

adeebshihadeh, 2025-07-25:
>     * [ ]  `BO_ 399` length 6* [ ]  `BO_ 399` length 7[x]

If this is real, it would be nice if the CANParser supported picking the right one with both present in the DBC.