commaai/opendbc #763: Many dbc files do not open in Vector CANdb++ Editor

seb43654, 2022-12-14:
Many .dbc files break the standard of the file format and hence do not open in Vector CANdb++, with an error being thrown for syntax. Given this program has the capability to easily view dbc files in a gui, and perform consistency/syntax checks it would be useful to support this. It also means many other applications which allow import of dbc files wont work with dbc files found in this opendbc repository.

collin80, 2022-12-14:
Do you know what it doesn't like? I've been using the opendbc files for a long time and haven't had a problem loading them. Lots of programs use them perfectly fine. So, maybe CANdb++ is just extra picky because it is a genuine Vector program.

adeebshihadeh, 2022-12-31:
Duplicate of #151

closed 2022-12-31 by adeebshihadeh