commaai/opendbc #869: CAN parser: assert signals exist

sshane, 2023-06-30:
We already assert messages exist, but signals are unchecked. This was the cause of https://github.com/commaai/openpilot/issues/28404, `CRUISE_INACTIVE` inside `SCC_CONTROL` doesn't exist, but it never complained or gave a warning before. @deanlee see any reason not to check signals too?

sshane, 2023-06-30:
Related to https://github.com/commaai/opendbc/issues/500

adeebshihadeh, 2023-06-30:
This should absolutely have an assert.

sshane, 2023-06-30:
The signal option name is passed all the way to the cpp code, which only loops through the DBC signals to find any matches, it doesn't check it finds one

https://github.com/commaai/opendbc/blob/e51a78721d99eac223f02cd9b993824712855952/can/parser.cc#L144-L150

deanlee, 2023-06-30:
how about check it in cython? https://github.com/commaai/opendbc/pull/870

closed 2023-07-01 by sshane