commaai/opendbc #913: CANParser: update_strings returns latest state when passed empty list

fredyshox, 2023-08-08:
`CANParser::update_strings` returns most recently updated addresses, when called with an empty list. 

Is it expected behavior? Shouldn't it return empty list of addresses, since nothing was updated?

Snippet showcasing this behavior:
```py
parser: CANParser = ...

addr = parser.update_strings(cans)
# addr = {...}
addr = parser.update_strings([])
assert len(addr) == 0
# ^ this assertion fails
```

sshane, 2023-08-08:
Yeah, some confusing interaction with the timing variables in parser.cc is going on. We can choose not to update the `vals` in `query_latest` if the last_ts is 0, but unsure if anything relies on that behavior

deanlee, 2023-09-05:
>  We can choose not to update the vals in query_latest if the last_ts is 0, but unsure if anything relies on that behavior since last_ts not being 0 is an explicit condition of skipping updating the values: 

currently, CANParser::__init__ relies on calling update_strings([]) to initialize vl & vl_all.

closed 2024-11-02 by sshane