commaai/opendbc #968: CANParser: invalid counter counter wraps around

sshane, 2023-11-04:
I assume the counter checks are intended to match between opendbc and panda, so while testing I noticed a mismatch as well as a bug:

panda clips the invalid counter counter to 5, but opendbc lets it wrap around no problem since it's a `uint8_t`:

https://github.com/commaai/opendbc/blob/562628092910903dab209bd6bfa4e16807f6b1bd/can/parser.cc#L80

```python
0x201 COUNTER FAIL #253 -- 9 -> 9
opendbc/can/parser.cc: 0x201 message checks failed, checksum failed 0, counter failed 1
ret.gas 63427.0 safety gas 17731
0x201 COUNTER FAIL #254 -- 9 -> 4
opendbc/can/parser.cc: 0x201 message checks failed, checksum failed 0, counter failed 1
ret.gas 63427.0 safety gas 17731
0x201 COUNTER FAIL #255 -- 4 -> 15
opendbc/can/parser.cc: 0x201 message checks failed, checksum failed 0, counter failed 1
0x201 COUNTER FAIL #2 -- 9 -> 1
ret.gas 23536.0 safety gas 17731
0x201 COUNTER FAIL #3 -- 1 -> 13
ret.gas 107119.0 safety gas 17731
0x201 COUNTER FAIL #4 -- 13 -> 9
ret.gas 57396.0 safety gas 17731
```

sshane, 2023-11-04:
@deanlee if you find this fun :slightly_smiling_face:

deanlee, 2023-11-05:
I'm afraid that I can't fix this properly. All I can think of at the moment is using a `counter_fail = std::min(MAX_BAD_COUNTER, counter_fial+1) `. But I'm not sure if it can really solve the problem.

sshane, 2023-11-06:
I think we want to match panda behavior here, so that is correct.

adeebshihadeh, 2023-11-06:
yeah let's just clip to something like uint8t_max

sshane, 2023-11-06:
panda clips to 5 (max bad counters), so there will be some mismatch there as well.

closed 2023-11-18 by sshane