commaai/opendbc #1000: Kia: unknown Elect_Gear_Shifter value 9

maximeborges, 2024-01-18:
Sometimes we see a CAN message for `Elect_Gear_Shifter` with a value of 9 on our Kia e-Niro.
It is supposed to be defined as follow:
https://github.com/commaai/opendbc/blob/40d9c723d48496229fecc436046538a53af19c11/hyundai_kia_generic.dbc#L1519-L1520

And the values:
https://github.com/commaai/opendbc/blob/40d9c723d48496229fecc436046538a53af19c11/hyundai_kia_generic.dbc#L1658

With this current definition I would expect the max value to be 8, is that correct?

Anyone already seen this? Otherwise I'll try to do some tests when I get access to one of those vehicle.

sunnyhaibin, 2025-03-12:
@maximeborges Are you still looking into this? Does this happen when you were trying to put into other forward gears?

maximeborges, 2025-03-12:
I don't have easy access to those cars and didn't try spending time trying to diagnose where that comes from yet. I might have some time next month that I can allocate to that