commaai/opendbc #1092: Chrysler: reliable fuzzy fingerprinting

sshane, 2024-08-20:
Same requirements as https://github.com/commaai/openpilot/issues/31052!

JIBSIL, 2025-01-08:
Is this bounty still open? If so I'd like to pick up where https://github.com/commaai/openpilot/pull/33000 left off

ReyNeill, 2025-12-03:
Bored, so I opened yet another PR (3 counting) on bounties with Open status:

Opened PR https://github.com/commaai/opendbc/pull/2928: Chrysler fuzzy fingerprinting now matches platform
part numbers (revision-agnostic) across cluster/ABS/radar/EPS and adds Chrysler-specific tests. Verified with
python -m pytest opendbc_repo/opendbc/car/chrysler/tests and python -m pytest opendbc_repo/opendbc/car/tests/
test_fw_fingerprint.py

PS: you guys need to address the community more efficiently I even offer myself to

sujeito-operator, 2026-08-28:
Opened #3709 for this.

Short version of the finding: Chrysler FW versions are Mopar part numbers — an 8 character base part number plus a two letter revision suffix — and the revision is bumped for running changes that don't change the platform. Matching on the base part number takes the fuzzy lookup from 516 exact versions to 380 part numbers while leaving the number of part numbers shared between platforms unchanged at 3, so it costs no specificity. On the checked-in database, a car running an unseen revision of a known part goes from 0/10 platforms fingerprinted to 10/10, with no cross-platform false matches.

Validated offline against the FW database and the test suite only — I don't have a car, so there's no route to attach.

sujeito-operator, 2026-08-29:
Withdrawing my claim on this bounty and closing #3709, so the board is not showing two live PRs for one row.

**#3520 (@flowmar47, 2026-07-04) is the one to review.** It predates mine by eight weeks, it is `mergeable`/clean with all twelve checks green including the `car diff` replay over 40 Chrysler segments, and on the one design point where the two PRs differ it is right and mine is wrong: it keeps `Ecu.eps` in the platform-code set. I had dropped `eps` because it appears in `FUZZY_EXCLUDE_ECUS`, but that list is applied only in the generic matcher (`fw_versions.py:73`) and never to the brand hook (`fw_versions.py:164-165`) — Hyundai's own `PLATFORM_CODE_ECUS` is `[fwdRadar, fwdCamera, eps]`, i.e. entirely ECUs from that same exclude list. Parsing `chrysler/fingerprints.py`, `eps` is on 10 of 10 platforms against `abs` on 8 of 10, so dropping it costs a signal for nothing. Reasoning in full on #3709.

The cross-brand check I posted on #3709 may still be useful when reviewing #3520: the `[A-Z0-9]{8}[A-Z]{2}$` Mopar shape matches Chrysler 519/519 and zero firmware strings in any of ford, honda, hyundai, mazda, nissan, subaru, toyota or volkswagen — so it is safe as a Chrysler-only matcher and there is no generic version of it worth writing.

ping-dev-ui, 2026-09-02:
Opened #3717 for this. It fuzzy matches on Mopar part numbers (revision letters ignored) and tolerates one platform-code ECU with an unseen part number, which is what a new model year usually changes. The PR description has the false-positive analysis on the current FW database and the tests that enforce it. CI is green.

closed 2026-09-15 by adeebshihadeh