commaai/opendbc #1097: TSSP reacts poorly to brake commands when creeping

haraschax, 2024-08-20:
### Describe the bug

see title

### Which car does this affect?

Toyota 2017 prius and TSSP rav4

### Provide a route where the issue occurs

NA

### openpilot version

master

sshane, 2024-11-13:
@haraschax what did this issue originally apply to? We also no longer support TSS-P long, so might be able to close

haraschax, 2024-11-13:
Toyota 2017 prius and TSSP rav4

sshane, 2024-11-13:
What does "reacts poorly" mean exactly?

haraschax, 2024-11-13:
Oh, when creeping small negative accel requests would not be honored, and it would continue to creep despite negative accel request until negative accel request became quite big. Forgot exact numbers.

adeebshihadeh, 2025-07-25:
@sshane is this still an issue?