commaai/opendbc #1102: Civic: permanent LKAS fault when jerking wheel

sshane, 2024-08-20:
Could be a diff of torque, max torque, or angle rate

https://connect.comma.ai/f299d0a03c4e965d/1685509313071/1685509343277

https://connect.comma.ai/f299d0a03c4e965d/1685509362131/1685509446705

https://connect.comma.ai/f299d0a03c4e965d/1685509216391/1685509226635

mvl-boston, 2025-07-27:
> If that helps, and the limits aren't very onerous, we may just apply them to all Honda and resolve this as well.

All cars with this issue reported are Bosch A.   So whatever fix is found, I don't think it needs to be expanded beyond Bosch A.

vanillagorillaa, 2025-07-27:
> > If that helps, and the limits aren't very onerous, we may just apply them to all Honda and resolve this as well.
> 
> All cars with this issue reported are Bosch A. So whatever fix is found, I don't think it needs to be expanded beyond Bosch A.

No, this is incorrect. Shane's initial report, and my follow up investigations both come from Nidec Civics.

mvl-boston, 2025-07-27:
Ok, haven't seen it on B or C yet.  Drivers didn't like the jerk shutoffs I
made global in earlier attempts to fix this as they liked the support of
LKAS even while manual steering.  So I would still recommend limiting to
only impacted car models.