commaai/opendbc #1108: OEM driver alert is activated when not touching steering wheel

laphlaw, 2024-08-20:
### Describe the bug

I can’t seem to find the rhyme or reason, but every so often I get an alert from my OEM system saying that no driver hands are detected on the steering wheel

Part of the appeal of buying a comma was the promise that I wouldn’t have to touch the wheel

### Which car does this affect?

2022 Hyundai Ionic 5

### Provide a route where the issue occurs

02bf13f6ee29bc3e|2023-12-21--14-28-33--0

### openpilot version

0.9.5

### Additional info

There’s also another segment that it happened on:
02bf13f6ee29bc3e|2023-12-21--15-57-33--0

At 4:07pm pacific

laphlaw, 2024-10-10:
apologies for the delay, I thought the files were auto uploaded.
Here is where the issue happens:
02bf13f6ee29bc3e/0000019a--4648cc1b15/4

jyoung8607, 2024-10-10:
This probably has some overlap with #1107, and may be an outright duplicate.

@sunnyhaibin did we ever identify a working mitigation for these cars? If nothing else, we could simulate driver input toward the stock camera (see the VW EA mitigation in #24711) but that's not exactly Plan A.

laphlaw, 2024-10-10:
I read through the issue that you linked. It seemed like in that one it is only happening when they plug in the comma AFTER starting the car. My unit is always plugged in, I never unplug I it. But yeah, the symptoms seem to be the same— I get the driver grasp warning at seemingly random times

jyoung8607, 2024-10-10:
> apologies for the delay, I thought the files were auto uploaded. Here is where the issue happens: 02bf13f6ee29bc3e/0000019a--4648cc1b15/4

Unfortunately only some of the logs are uploaded, and we don't have the one for 12 minutes in.

[image]

> I think the actual OEM driver alert ("USER IS NOT DETECTED") comes on around 12:17, but I can't say for sure

It's going to be really important to know the incident timestamp as closely as possible. If this happens frequently, do something like wave your hand in front of the camera, or use the bookmark function in the openpilot UI.

If you haven't used the bookmark function before: touch the screen, when the bar appears on the left with the Settings button at the top, there's a bookmark icon at the bottom. Touch that. There won't be a visible reaction, but it leaves a timestamped trace for us in the logs that's easy to track down later.

laphlaw, 2024-10-14:
ok I repro'd again with this route:
02bf13f6ee29bc3e/000001ad--6b14a6336c/12

I added a bookmark with the UI, let me know if you don't see the bookmark. it should be within +/- 2 minutes of 16:50.

laphlaw, 2024-10-25:
@jyoung8607 any update here?