commaai/opendbc #1109: Is support available for “Hyundai Genesis GV70 2021 2.5T”?

cloudran, 2024-08-20:
### Describe the bug

Describe the bug
Is support available for “Hyundai Genesis GV70 2021”?
At comma 3x, the vehicle is not recognized and starts in dashcam mode.
So I tried extracting my car's fingerprint and firmware information and adding them to value.py, but it still goes into dashcam mode.
I am sharing the fingerprint and firmware information I extracted for reference, and if you have any information you need, please let me know.

Which car does this affect?
Hyundai Genesis GV70 2021

Provide a route where the issue occurs
81a6ddf7edcd84f4|2023-10-14--07-45-08

openpilot version
0.9.4

Additional info
Fingerprint information
48: 24, 64: 32, 74: 32, 96: 32, 111: 8, 160: 24, 181: 32, 234: 24, 256: 32, 282: 16, 293: 16, 298: 16, 314: 32, 320: 8, 325: 32, 352: 16, 362: 32, 373: 24, 416: 32, 426: 16, 432: 32, 437: 32, 442: 24, 463: 8, 480: 16, 485: 16, 490: 32, 506: 32, 512: 8, 517: 24, 549: 16, 608: 32, 688: 32, 698: 32, 794: 32, 837: 8, 842: 32, 879: 8, 895: 8, 896: 8, 897: 8, 898: 8, 899: 8, 900: 8, 923: 8, 934: 8, 944: 8, 945: 8, 946: 8, 961: 8, 962: 8, 1040: 8, 1041: 8, 1042: 8, 1043: 8, 1044: 8, 1046: 8, 1047: 8, 1048: 8, 1049: 8, 1050: 8, 1051: 8, 1052: 8, 1057: 8, 1058: 8, 1062: 8, 1063: 8, 1065: 8, 1066: 8, 1070: 8, 1071: 8, 1076: 8, 1090: 8, 1092: 8, 1107: 8, 1141: 8, 1167: 8, 1174: 8, 1187: 8, 1204: 8, 1208: 8, 1209: 8, 1210: 8, 1214: 8, 1215: 8, 1240: 8, 1247: 8, 1264: 8, 1266: 8, 1267: 8, 1278: 8

Vehicle firmware information
{'carParams': {'alternativeExperience': 1,
               'autoResumeSng': True,
               'carFingerprint': 'mock',
               'carFw': [{'address': 1975,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'cornerRadar',
                          'fwVersion': b'\xf1\x00JK1 BCW RR 1.00 , 1.01 (s\x94@ \x03ISH',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1983,
                          'subAddress': 0},
                         {'address': 2000,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00JK1_ SCC -----      1.00 1.02 99110-AR100         ',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 2008,
                          'subAddress': 0},
                         {'address': 1971,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'hvac',
                          'fwVersion': b"\xf1\x00JK1   97255-AR011CONTROL ASS'Y-DATC  1.04JK1 DATC(-)1.5       ",
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1979,
                          'subAddress': 0},
                         {'address': 1988,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00JK1 MFC  AT KOR LHD 1.00 1.04 99211-AR100 210204',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1996,
                          'subAddress': 0}],
               'carName': 'mock',
               'carVin': '00000000000000000',
               'centerToFront': 1.350000023841858,
               'communityFeatureDEPRECATED': False,
               'dashcamOnly': False,
[... 154 lines cut ...]
               'safetyModelPassiveDEPRECATED': 'silent',
               'safetyParamDEPRECATED': 0,
               'startAccel': 0.0,
               'startingAccelRateDEPRECATED': 0.0,
               'startingState': False,
               'steerActuatorDelay': 0.0,
               'steerControlType': 'torque',
               'steerLimitAlert': False,
               'steerLimitTimer': 1.0,
               'steerRateCostDEPRECATED': 0.0,
               'steerRatio': 13.0,
               'steerRatioRear': 0.0,
               'stopAccel': -2.0,
               'stoppingControl': True,
               'stoppingDecelRate': 0.800000011920929,
               'tireStiffnessFactor': 0.0,
               'tireStiffnessFront': 1000000.0,
               'tireStiffnessRear': 1000000.0,
               'transmissionType': 'unknown',
               'vEgoStarting': 0.5,
               'vEgoStopping': 0.5,
               'wheelSpeedFactor': 1.0,
               'wheelbase': 2.700000047683716},
 'logMonoTime': 345529893018,
 'valid': True}

sunnyhaibin, 2025-03-12:
@Lindorak @cloudran @wyatt-pontius is this still an issue? 2022-24 GV70 2.5T is supported in `master` for a while. If you are still having issues, please report, otherwise we should be ready to close this. (cc @jyoung8607 @sshane )

wyatt-pontius, 2025-04-05:
> @Lindorak @cloudran @wyatt-pontius is this still an issue? 2022-24 GV70 2.5T is supported in `master` for a while. If you are still having issues, please report, otherwise we should be ready to close this. (cc @jyoung8607 @sshane )

Hey @sunnyhaibin, vehicle still shows as dashcam mode (car unrecognized)

wyatt-pontius, 2025-04-05:
cc @jyoung8607 @sshane

wyatt-pontius, 2025-04-05:
> @wyatt-rematter 
> 
> I haven't made any progress since the last time I've posted. Been personally dealing with life so I've put this on the back burner. 
> 
> A couple questions:
> - Which country are you located in?
> - Which trim of the 3.5 do you have? There is the higher trim and the lower trim. 
> 

@Lindorak I am in the US, trim is 3.5 sport prestige (higher trim)

wyatt-pontius, 2025-05-17:
@sunnyhaibin @jyoung8607 @sshane just bumping this open, in case it's of interest. I have the 2022 GV70 3.5 - haven't done too much more troubleshooting on this since the thread is dead