commaai/opendbc #1117: Kia Niro 2019 - "Check FCA" warning when engaging openpilot

DerGue69, 2024-08-20:
### Describe the bug

I have a Kia Niro EV 2019 (EU - Austria - 1st Edition). A "Hyundai H" Harness is my connector.
After installin stock OP my car was not recognized. I had to add values to values.py from my cars fingerprint (see additional info)
(I thought My Niro EV is already compatible with OpenPilot)

After that OP started working with my car. But everytime I activate OP I get a "Check FCA ( Forward Coll. Avoidance Assist)" warning. 

I recorded a route and made public.
https://cabana.comma.ai/?route=6111bd7df5043394%7C2022-07-20--14-47-58&exp=1689861829&sig=9eylZ8Csg1lvrMr6yneTzSZbSC7qN5ZjymwZ2jOTkVM%3D&max=-1&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F6111bd7df5043394%2Fc7e80e23a0c545d687808655907ee3ee_2022-07-20--14-47-58

Please help me to get rid of this error ;)

[image]

### Which car does this affect?

Kia Niro EV 2019 (EU - Austria - 1st Edition)

### Provide a route where the issue occurs

6111bd7df5043394%7C2022-07-20--14-47-58

### openpilot version

0.8.14

### Additional info

[image]

'carFingerprint': 'KIA NIRO EV 2020',
               'carFw': [{'address': 2004,
                          'ecu': 'eps',
                          'fwVersion': b'\xf1\x00DE  MDPS C 1.00 1.04 56310Q4100\x00 4DEEC104',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 2000,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00DEev SCC F-CUP      1.00 1.02 96400-Q4000         ',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 1988,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00DEE MFC  AT EUR LHD 1.00 1.03 95740-Q4000 180821',
                          'responseAddress': 0,
                          'subAddress': 0}],

sunnyhaibin, 2025-03-12:
@DerGue69 Can you test on the latest `master` and see if you are still having this issue?

DerGue69, 2025-03-13:
> [@DerGue69](https://github.com/DerGue69) Can you test on the latest `master` and see if you are still having this issue?

I will test on latest master of stock Openpilot in the next couple of days. 
Let me know if you meant Sunnypilot master - then I will test the branch as well.

adeebshihadeh, 2026-02-08:
@DerGue69 did you end up testing it?

DerGue69, 2026-02-08:
> [@DerGue69](https://github.com/DerGue69) did you end up testing it?

Thanks for the reminder. I installed 0.10.0 master today.
Unfortunately same FCA error. No change. I will go back to my fork.

This route is presved and public for investigation for you:
6111bd7df5043394/00000001--bd7a133722/0

[image]
[image]
[image]