commaai/opendbc #1125: GM: Cruise Fault at standstill

morrislee, 2024-08-20:
### Describe the bug

Some cases, where I like to use stock ACC or OpenPilot ACC to hold my brake at a stop light, occasionally would trigger Cruise Faulted red alert.

I am able to replicate this on master, master-ci and other forks as well. Happens both on stock and op ACC.

[image]

### Which car does this affect?

2020 GMC Sierra 1500

### Provide a route where the issue occurs

dc7716b32bf25574|2023-12-22--21-09-03--5

### openpilot version

0.9.6

jyoung8607, 2025-07-07:
@1454 - can you open a separate issue for that? The original issue seems specific to standstill, and you're slowing down while engaged. We can still look at it, but the cause may be different.

@morrislee - if you can still replicate this, would you be so kind as to upload a couple more examples? Unfortunately the ones you shared have expired. If you mark them Preserved in comma connect, we'll have them available for longer.

1454, 2025-07-07:
I can. I was waiting to hear back from Tim if he remembers the one he opened for me back last year at some point. I tried searching but didn't return that specific one. I just didn't want to open two issues for the same thing.

morrislee, 2025-07-07:
> [@1454](https://github.com/1454) - can you open a separate issue for that? The original issue seems specific to standstill, and you're slowing down while engaged. We can still look at it, but the [...]

this is still happening, but its far in between as its really random to reproduce. Has to be just coming to a standstill and hitting the resume button not the set button to trigger this. I had one about a week ago, but its too far back and logs are no longer available