commaai/opendbc #1143: Subaru: Forester 2022 support

AlterTheEnding, 2024-08-20:
### Describe the bug

I recently got the comma three and my device will only work in dashcam mode on my 22 Subaru Forester Wilderness. I’m new to all this so I hope I report all this correctly, I apologize in advance. 

### Which car does this affect?

Subaru Forester Wilderness 2022

### Provide a route where the issue occurs

8db3f557a1e70a1a|2022-11-26--19-49-23

### openpilot version

Master-ci

### Additional info

I first tried the open pilot release dashcam mode only then the Subaru communities fork/branch (sorry new here) and that was dash cam mode only then read the help page for fingerprinting and tried master-ci version and still dashcam mode only. I truly appreciate your help!

martinl, 2024-08-28:
> So did this work? Thinking about getting a comma for my 2024 Forester, but don’t want to spend that much if it is only a dashcam. Glad to help where I can with development, though (testing, etc.).

Angle-based models like Forester 2022-2023 have support in a separate branch but are still waiting for review before getting merged to upstream. You can always return the comma device within 30 days no questions asked if you are not happy with it

jasonwbarnett, 2025-04-27:
@martinl With the relatively recent release of 0.9.6 with support of Toyota's angle based steering control API, does that make Subaru Forest 2022-2024 more likely to be implemented? I have a 2022 Forest Wilderness Edition and I'd love to buy 3x and use comma.