commaai/opendbc #1147: Rumbling while collaboratively steering at low speeds

voximity, 2024-08-20:
### Describe the bug

At low speeds (around 15 MPH), my steering wheel will "rumble" if I collaboratively steer alongside lateral input from openpilot. To be precise, this is not the kind of vibration/rumbling that Hondas give the driver to indicate a warning (like lane departure), but almost feels as if I am going over a rumble strip. More accurately, it sounds like I'm fighting the torque actuator -- or something.

### Which car does this affect?

Honda CR-V 2020

### Provide a route where the issue occurs

e4b43f41460a7547/00000001--02564f7b2d/1

### openpilot version

27c26c2 (master-ci)

### Additional info

I initially encountered this issue on Sunnypilot when using its separated LKAS / ACC feature. My CR-V's stock ACC does not allow me to set the cruise below 25 MPH, so I was able to reproduce this on stock OP using OP long and setting my max speed to 15 MPH. In the provided route, the issue happens a lot as I zig-zag through a street near the very end (immediately preceding the bookmarks at the end).

*Possibly* related to commaai/opendbc#1112, except this feels *and* sounds like a rumble, rather than just being an electric humming sound.

jyoung8607, 2025-07-27:
The [2020 Honda CR-V owner's manual](https://cdn.dealereprocess.org/cdn/servicemanuals/honda/2020-crv.pdf) shows three causes of tactile warnings:
* Lane Keeping Assist System
* Road Departure Mitigation
* Driver Attention Monitor

For LKAS and RDM, I'm not immediately able to find anything interesting coming from the camera, and we're outside the documented speed envelope for either of them to activate.

Driver Attention Monitoring (just an sensor/algorithm thing) may not be done by the camera, so we may not see it in CAN logs. On at least one other manufacturer with a similar algorithm, it's done in the CAN gateway. We could have a similar situation with Honda, it could even be done within the EPS itself. But we're also outside its documented speed envelope.

We're operating your EPS well below its documented speed support, so by definition we can encounter undocumented behavior. There could be some sort of low speed driver takeover warning built into the EPS firmware.

Questions:
* Do any other Honda CR-V owners have this behavior?
* Do you have Driver Attention Monitor turned on? There's a tactile-only setting. Try turning it off.
* We could test with reapplication of #2104

jyoung8607, 2025-07-27:
Discord discussion turned up the same thing on Honda City, which isn't upstream yet but had to slightly lift its min steer speed for the same issue. The EPS is trying to do us a solid because we're leaving its control envelope, so we'll take the hint.

jyoung8607, 2025-07-28:
This was reopened because the change in #2570, while valid and appropriate, doesn't actually prevent openpilot from trying to steer. Correctly fixing this will require actual minSteerSpeed handling in the Honda port, which is a larger undertaking.

bvleonov, 2026-09-10:
I'm having a similar issue in my Kia Niro, so it seems like this affects different brands. Report here: https://github.com/commaai/opendbc/issues/3685#issue-5144010979