commaai/opendbc #1153: Subaru: camera FW query isn't complete on startup

sshane, 2024-08-20:
Brought to our attention from the FW bot, many Subarus return different results for the same ECU and request:

```python
1c451e9ed3c48243|2024-01-15--16-10-38:
[('ecu', 'fwdCamera'), ('fwVersion', b'\x00\x00e\x9a\x00\x00\x00\x00\x00\x00'), ('address', 1927), ('subAddress', 0), ('responseAddress', 1935), ('request', [b'"\xf1\x82'])
vs
1c451e9ed3c48243|2024-01-15--20-36-22:
[('ecu', 'fwdCamera'), ('fwVersion', b'\x00\x00e\x9a\x00\x1f@ 1\x00'), ('address', 1927), ('subAddress', 0), ('responseAddress', 1935), ('request', [b'"\xf1\x82']) (edited) 
```

Race condition on startup?

Also looks like we've been adding them to the database:

```
    (Ecu.fwdCamera, 0x787, None): [
      b'\x00\x00df\x1f@ \n',
      b'\x00\x00d\x95\x00\x00\x00\x00',
      b'\x00\x00d\x95\x1f@ \x0f',
      b'\x00\x00d\xfe\x00\x00\x00\x00',
```

Bounty
---

After 10 seconds our Outback's fwdCamera returns the full non-zero-suffixed FW. b'\x00\x00eJ\x00\x00\x00\x00\x00\x00' -> b'\x00\x00eJ\x00\x1f@ \x19\x00'. There is another FW version-related data ID (DID) on 0xf100 when using read data by ID, can we use that instead, or is it the bootloader software version or similar?

Bounty requirements:
- Our existing 0xf182 RDBI query returns the complete FW version every time, even right after startup
- OR using our [commaCarSegments](https://blog.comma.ai/096release/#commacarsegments) database, you can verify that we can use another ECU or DID. There's a EyeSight Monitor ECU that we can query that returns the physical part number and software version for example, but it's a HUD ECU.
  - We can use another ECU or DID if you verify that it has enough information to identify the API of the car reliably -- if it has LKAS and ACC we can control -- and we don't fingerprint on cars without the complete API to control the car.

rhass, 2024-08-20:
Just thinking out loud here.. but didn't the third-gen EyeSight system add a third camera with a different FoV (and possibly different object detection model)? Perhaps the returned values pertain to the respective cameras; outter cameras vs middle camera?

bravochar, 2024-09-13:
I wasn't experiencing this at first with my 2023 Ascent; I was getting back `b'\x05!\x08\x1dK\x05!\x08\x01/'` fairly consistently back in July. Now I am getting `b'\x05!\x08\x1dK\x00\x00\x00\x00\x00'`, which matches the first 5 bytes. This only happens when the Comma is already running at the time the car starts. If it has shut off completely (due to unplugging or time between starts), I then get the full 10 bytes.

bravochar, 2024-09-13:
Is it feasible to mask the firmware version for matching purposes? 5 bytes is more than some of the other ECU's full versions. Given that my Ascent matches the one [here](https://github.com/commaai/openpilot/pull/31545) for all 10 bytes, I'm hoping the case can be made that the first 5 bytes are enough. I'll take a look at the commaCarSegments and see if there's data there to support the theory.

greylot, 2025-07-12:
**Retry Mechanism Method**: 
Query the 0xf182 DID at 2-second intervals up to 5 times to retrieve the full firmware (FW) version. The query is considered successful if the last bytes of the returned version are not zero.

**Why Suitable?**:
- - Covers a 10-second delay (5 x 2 seconds).

- - Maintains the use of the existing 0xf182 DID and fwdCamera ECU.

- - Validation with commaCarSegments prevents false positives.

**Advantages**: Simple, adaptable, requires no system changes.

**Implementation Code**

Sample code written in Python (using the python-uds library):
```
from uds import Uds
import time

def query_firmware_version(ecu_address=0x787, max_attempts=5, delay=2):
    """
    Queries the 0xf182 DID and retrieves the full FW version.
    """
    try:
        client = Uds(interface="can", tx_id=ecu_address, rx_id=0x78f)
        client.send([0x10, 0x03])  # Extended Diagnostic Session
        for attempt in range(max_attempts):
            response = client.send([0x22, 0xf1, 0x82])  # 0xf182 query
            fw_version = response[3:]  # Extract FW version
            if not fw_version.endswith(b'\x00\x00\x00\x00'):
                print(f"Full FW version: {fw_version}")
                return fw_version
            print(f"Attempt {attempt + 1}: Incomplete version: {fw_version}")
            time.sleep(delay)
        raise Exception("Failed to retrieve full FW version")
    except Exception as e:
        print(f"Error: {e}")
        return None
    finally:
        client.disconnect()

def verify_firmware_in_database(fw_version, database):
    """
    Verifies if the FW version supports LKAS/ACC.
    """
    return fw_version in database and database[fw_version].get("has_lkas_acc", False)

# Example database
comma_car_segments = {
    b'\x00\x00e\x9a\x00\x1f@ 1\x00': {"has_lkas_acc": True},
    b'\x00\x00df\x1f@ \n': {"has_lkas_acc": True},
    b'\x00\x00e\x9a\x00\x00\x00\x00\x00\x00': {"has_lkas_acc": False}
}

# Usage
if __name__ == "__main__":
    fw = query_firmware_version()
    if fw and verify_firmware_in_database(fw, comma_car_segments):
        print("Vehicle supports LKAS and ACC")
    else:
        print("Vehicle does not support LKAS/ACC or unknown version")
```

**Test Plan Vehicles**: 

At least 3 Subaru Outback models (e.g., 2020, 2022, 2024).

**Hardware**: Tactrix OpenPort 2.0 or similar CAN interface.

**Scenarios**:

**Cold Start**: Turn off the vehicle, wait 1 minute, start, and query immediately.

**Warm Start**: Run the vehicle for 5 minutes, turn off, restart, and query.

Note:
- - Alternatively, another ECU or DID (e.g., 0xf100, EyeSight Monitor ECU) can be used, but LKAS/ACC support must be verified with the commaCarSegments database.

- - False positives (vehicles without LKAS/ACC being shown as compatible) should be prevented.

amazing2000zame-hub, 2026-11-15:
Submitted verified fix and unit tests in PR #3957 resolving incomplete startup EyeSight fwdCamera FW matching.