commaai/opendbc #1155: Mazda: 2024 CX-5 Signature OEM Front Camera Sensor Failure Error

DouglasSherk, 2024-08-20:
### Describe the bug

This appears to be exactly the same as commaai/openpilot#25525 but on a 2024 CX-5 Signature. Copying and pasting that report as it's still exactly accurate:

> When driving with OP active, front camera sensor error shows up on the OEM instrument cluster
> 
> similar error to https://vf.vin/static/uploads/images/sr2/mazda/2022/02/18/2018-mazda-cx-5-jm3kfbdm5j0335098-img-13.jpg
> 
> Error goes away with disengage/rengage OP
> 
> No errors seen on C3 unit and self drive does not disengage and does not seem affected

The rest of the filer's description is accurate too.

I have tried on:
* Stock OP 0.9.5
* A custom version of 0.9.5 I made with @Jafaral's [mazda-tja](https://github.com/Jafaral/openpilot/tree/mazda-tja) commits rebased onto it: https://github.com/DouglasSherk/openpilot/tree/mazda-tja

### Which car does this affect?

2024 Mazda CX-5 Signature

### Provide a route where the issue occurs

60ed2cf2b490292c|2023-11-21--17-47-15--12

### openpilot version

0.9.5

### Additional info

I noted that:

> The lane lines are no longer present on the lane-keeping assist dashboard, so I'm guessing it has something to do with that. When I unplug the C3 but keep the harness in-place, everything goes back to working stock, so it's not a problem with the wiring.

@Jafaral suggested the following:

> Lane lines going away is just a result of the camera error, and not the cause of it., most of the info you see on the dashboard are just decorations really.
> CAN data might reveal more info

Hence I collected the route and filed this report.

I don't think trims lower than Signature have Traffic Jam Assist, which appears to be the source of the problem. That might be why no one else with a 2022-2024 CX-5 has experienced this issue.

I'm a total newbie at OP, but a competent developer otherwise if that's helpful.

ljbeato, 2025-01-06:
I just installed my new comma 3x on my 2024 CX-5 Turbo Premium trim. I believe it is running OpenPilot 0.9.7.
I started getting this error several times today. The car seems to steer fine, but the error keeps popping up. Any ideas as to what is causing it? 
Thx

Connor-Knabe, 2025-06-04:
@DouglasSherk did you ever come up with a good solution for this? 🤞

Connor-Knabe, 2025-06-19:
@rnabil0 sunnypilot staging-new branch seems to fix the issue