commaai/opendbc #1169: GM: Very slow acceleration when resuming from a stop or slow roll

Verylukyguy, 2024-08-20:
### Describe the bug

Very slow acceleration when resuming from a stop or after slowing to a near stop for a lead car to turn; it does get normal at around 30 mph.

### What hardware does this issue affect?

comma two

### Which car does this affect?

GMC Acadia 2018

### Provide a route where the issue occurs

e38a02cae1ef1e0c|2022-04-15--07-43-35--3

### openpilot version

Master 0.8.14 @ 1fa1c44

### Additional info

This has been an ongoing issue from the beginning for the GM port.

@Twilsonco has fixed this in his fork and has even added a toggle for Normal/Sport/Eco/Creep acceleration modes.

Verylukyguy, 2025-03-25:
@sshane Does merging #951 solve this?

jyoung8607, 2025-07-05:
Should be resolved by #951.

Verylukyguy, 2025-07-06:
I am still experiencing this issue, this did not resolve it.

jyoung8607, 2025-07-07:
> I am still experiencing this issue, this did not resolve it.

Can you share a recent route on current openpilot?

sshane, 2025-07-07:
As noted in that PR, it does not change any behavior. We still have to change the max value dependant on the car.

jyoung8607, 2025-07-07:
> As noted in that PR, it does not change any behavior. We still have to change the max value dependant on the car.

Ah, I knew the signal-width issue was sorted, but I missed that we weren't making use of the additional width yet.

@Verylukyguy We still do need routes demonstrating the issue as it stands.

Verylukyguy, 2025-07-15:
@sshane There are three resumes from a stop in this route df51b2195de66a9b/00000005--739b7b6c5a/0