commaai/opendbc #1173: [$300 Bounty] Toyota: cancel cruise without chime

adeebshihadeh, 2024-08-20:
The way openpilot cancels cruise control on most Toyotas has the side effect of a chime from the car and a momentary fault on the dash. When openpilot sends the cancellation message, it also sends a message to play a good chime to mask the fault chime that would play, so start by [removing that](https://github.com/commaai/openpilot/blob/fa62b9d3f9882ec33559ceb7377fa991be1d4fb5/selfdrive/car/toyota/toyotacan.py#L82). Once you remove the good chime, you should hear the fault chime, which we're trying to get rid of.

The Lexus IS uses a [different cancellation message](https://github.com/commaai/openpilot/blob/fa62b9d3f9882ec33559ceb7377fa991be1d4fb5/selfdrive/car/toyota/carcontroller.py#L94) than the rest of the Toyotas, so that's probably a good place to start.

$200 for a merged PR that meets this criteria:
1. no chime sound or even momentary fault display
2. reliable, i.e. always immediately cancels cruise
3. generic across all Toyotas

Feel free to open a PR once steps 1 and 2 are done to lock the bounty pending confirmation of 3.

thomasthinks, 2026-10-04:
## Proposal to close this bounty with an explanation

I'd like to argue this bounty is not achievable as specified, and should be closed. The acceptance criteria demand all three of: (1) no chime, no momentary fault display, (2) reliable, (3) generic across all Toyotas. After four years of attempts, every candidate mechanism fails at least one criterion on-car, and the failures are documented by the people who tried them.

**1. `CANCEL_REQ` in `ACC_CONTROL` (the current approach)** — the only reliable force-cancel, but the PCM treats a CAN-forced cancel as a fault and chimes. The `TWO_BEEPS` mask in `create_ui_command` doesn't make it silent; as cydia2020 pointed out in this thread, removing the mask (opendbc#3257) can't remove the car's own fault chime — that PR was closed unmerged. The TODO is still in `toyotacan.py` today: `# TODO: find the exact canceling bit that does not create a chime`.

**2. `PRE_COLLISION` STATE=3 spoof (openpilot#23404)** — the only approach that ever achieved silence, and it was closed by adeebshihadeh as "uses the AEB message in a way that isn't well understood and doesn't work in all conditions." On-car results: it doesn't work at zero CAN speed (2018 Prius TSS-P, cydia2020), it caused a cruise-control malfunction requiring a full car restart when brake-holding at standstill (2020 Corolla, nelsonjchen), it risks counter mismatches with the stock DSU/radar `PRE_COLLISION` stream on cars where the DSU stays plugged in (pd0wm), and it slightly applies the brakes (cydia2020). The author himself: "I don't know why this worked. Only toyota knows why." Fails criteria (2) and (3).

**3. Everything else has been tried on-car and disproven:**
- PCS messages `0x283`/`0x344` (wocsor's hint): sshane tested on a TSS2 Camry — mutes the stereo, does not disengage.
- `MINI_CAR=0`: 4 beeps plus a "NO CLOSE LEAD CAR" dash display — fails criterion (1).
- `ACC_CONTROL` bit 29, `PCM_CRUISE_2` bit 3, `GEAR_PACKET` neutral spam, `BRAKE_MODULE` variations: sshane tried them, no disengage.

**Root cause:** Toyota's PCM only cancels silently on genuine driver input (pedal/button). A cancel forced over CAN is indistinguishable from a fault to the PCM, and no reverse-engineered message has been found that decouples the two — not by the community, and not by comma's own team members who worked this thread.

If the maintainers want to keep pursuing silence, the bounty would need rescoping — e.g. TSS2-only, or accepting the current masking behavior. As written, with (1)∧(2)∧(3) all required, I believe this is exhausted with current knowledge, and I'd ask that it be closed per the close-with-explanation path.

cydia2020, 2026-10-04:
> I'd like to argue this bounty is not achievable as specified, and should be closed. 

As unfortunate as it is, I tend to agree. The best way to handle this would still be what I'm doing on my own fork, which is not to send cancel unless there's an actual need for it.

Currently, and most annoyingly, openpilot sends cancel on brake pedal actuation, which is unnecessary as the car would disengage regardless. If we only send cancel on fault (e.g. driver not paying attention/steering unavailable/service stopped/etc.) it would actually make sense to display a "Cruise Control unavailable" error together with the chime.