commaai/opendbc #1192: Revert fingerprint of Toyota Corolla 2021 Hybrid China version

0312birdzhang, 2024-08-28:
I made a pull a request 2 years ago, https://github.com/commaai/openpilot/pull/25026

However, this car does not have BSM when it leaves the factory, with this fingerprint added,  it causes some same car users not be able to use openpilot normally. Even if the BSM is installed later, there are some models of BSM that cannot be captured by openpilot, like my RV1.

Please delete this line, i'm very sorry.
https://github.com/commaai/opendbc/blob/3a8a8645f61e30ae61ee0f6458c1e34fe53979e5/opendbc/car/toyota/fingerprints.py#L566

I you need BSM error logs, i will provide later.

[image]

sshane, 2024-11-12:
Your car would still match via fuzzy fingerprinting if removed. We also detect BSM via the dynamic fingerprint, what exactly is the problem here? Going to close as this should be a PR

closed 2024-11-12 by sshane

0312birdzhang, 2024-12-03:
I have to set `ret.enableBsm` to False, otherwise the log will be full of the following errors.
https://github.com/commaai/opendbc/blob/651be58ac8cd8acaf2754905a78b0c187a99aa25/opendbc/car/toyota/interface.py#L103

[image]
(Yes, the screenshot is from a Xiaomi Mi Pad 5 running openpilot)

It has BSM, but it reads empty signals from `0x3F6`.

0312birdzhang, 2024-12-03:
> Can you provide a route with rlogs uploaded?

My C3x is on the way, will do it then.

0312birdzhang, 2024-12-05:
I would say you are right, removing the fingerprint does not resolve the BSM issues. I'm using a clone C3 now, and my C3x from comma is on the way, will do a upload via connect.comma.ai when i received.

0312birdzhang, 2025-01-23:
After rearranging the order of bsm modules and panda, bsm can be detected by openpilot.

[image]