commaai/opendbc #1296: Volkswagen B harness

jyoung8607, 2024-09-27:
**New Harness Checklist**

- [x] which car is this for? Volkswagen in general newer than MY2022, including MQBevo (but not MEB)
- [ ] schematic - Push to neo repo once confirmed working
- [ ] beta sent to tester
- [ ] in shop
- [ ] Added to openpilot docs

**Additional Info**

This harness is intended to work for newer MQB with the 2Q0 camera, and also MQBevo with the 5WA camera. The wiring between the two types isn't the same, but for our purposes, they are compatible enough to combine.

The male connector housing is `8W0 972 112 A` ([example source](https://www.ecstuning.com/b-genuine-volkswagen-audi-parts/housing/8w0972112a/)). It's pricey, but there also seems to be a Taobao seller.

The female connector housing has not been found on the market. @vanillagorillaa went so far as to purchase and disassemble the camera, and the PCB mount housing had no identifying markings. @lukasloetkolben has a 3D printable model.

While comma production is pending, a limited number of prototype kits are available [directly from Lukas](https://www.ebay.com/itm/395616600662).

**Harness Wire Mappings**

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/jyoun/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/jyoun/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#467886" vlink="#96607D">

Terminal | MQB 2Q0 (New, Tan) | MQBevo (Tan) | Harness Mapping
-- | -- | -- | --
1 | Extended CAN High | ADAS CAN High | CAN 0/2 High
2 | Extended CAN Low | ADAS CAN Low | CAN 0/2 Low
3 | Fusion CAN High |   | CAN 1 High
4 | Fusion CAN Low |   | CAN 1 Low
5 | Camera heater supply | Camera heater supply | Passthrough
6 | Camera heater ground | Camera heater ground | Passthrough
7 | Ground | Ground | Ground
8 | Switched 12V | Switched 12V | Switched 12V
9 |   |   |  
10 |   |   |  
11 |   | Fusion Ethernet High | Passthrough
12 |   | Fusion Ethernet Low | Passthrough

</body>

</html>

**Source Documentation**

[Camera WD - MQB 2Q0 (New).pdf](https://github.com/user-attachments/files/16752747/Camera.WD.-.MQB.2Q0.New.pdf)

[Camera WD - MQBevo.pdf](https://github.com/user-attachments/files/16752748/Camera.WD.-.MQBevo.pdf)

jyoung8607, 2024-10-30:
Some notes about more fallout from #1272:

* 2024 Volkswagen Taos also split, also on February 19, discovered from an end user. Don't need to revoke support because MY2024 was never officially added.
* 2024 Volkswagen Tiguan also split, also on February 19
* 2024 Volkswagen Arteon also split, on March 4, but looks like there might have been an actual gap in production
* Volkswagen Atlas and Golf had already refreshed to MQBevo with CAN-FD, along with Audi A3

In short, we don't have a harness for any current production USA market cars. We need the VW B harness pretty badly.

zekares, 2025-03-09:
I found this on aliexpress 

12 Pin 1-1534096-1 8E097211 2A

https://tr.aliexpress.com/item/1005002524085907.html

https://tr.aliexpress.com/item/1005004369605026.html

Does it work?