commaai/opendbc #1298: Bosch C Harness

vanillagorillaa, 2024-09-27:
- [x]  which car is this for? 2023 Honda Pilot, 2023 CR-V, 2023 Accord
- [x]  schematic - Push to neo repo once confirmed working
- [x]  beta sent to tester
- [ ]  in shop
- [ ]  Added to openpilot docs

Reported on Discord by user "Kevmaninc". Unknown connector part number. Will have to grab wiring diagram from Honda Techinfo. Unknown camera module. No marking on underside of camera. Will assign proper harness name once more information is obtained.

vanillagorillaa, 2025-03-08:
https://github.com/commaai/website/commit/01bb7f63ef87290a416773bb1867634f3a8b5d7a

closed 2025-03-08 by vanillagorillaa

shahrk, 2025-03-09:
> https://github.com/commaai/website/commit/01bb7f63ef87290a416773bb1867634f3a8b5d7a

Does this mean CRV 2023 should be supported now? The website still doesn't show that it's supported.

Kevmaninc, 2025-03-09:
That is correct, the CRV 2023 has the same support as Pilot 2023.

ba5714, 2025-04-03:
What is the latest fork for honda pilot 2023?

Kevmaninc, 2025-04-03:
> What is the latest fork for honda pilot 2023?

You should just be able to use the main OpenPilot as it should work now. However if you are interested in a SunnyPilot version then you have some options. 

VG Pilot Install:
installer.comma.ai/vanillagorillaa/Pilot_2023

Babyhamsta
installer.comma.ai/Babyhamsta/dev-c3

ba5714, 2025-04-03:
On the main open pilot, it doesn't work. Maybe a newer version is needed?

Kevmaninc, 2025-04-04:
Try the links I sent you.

ba5714, 2025-04-16:
installer.comma.ai/vanillagorillaa/Pilot_2023

This fork stopped working. Why? What is the replacement?

Thanks

Mr-Peter-Ng, 2025-04-17:
@ba5714 that link is being worked on & updated. It should be back up & running soon.

ba5714, 2025-04-17:
No, it's still not working.

Kevmaninc, 2025-04-17:
Are you using the main comma branch? You need to go to discord and check with the community.Sent from my iPhone

ba5714, 2025-04-17:
I'm trying to use this one.

installer.comma.ai/vanillagorillaa/Pilot_2023

Kevmaninc, 2025-04-18:
No, you need to use something else. Try the one below and choose your car
from the list.

installer.comma.ai/kevmaninc/dev-c3

smtermini, 2025-05-04:
So I just purchased the Bosch C Harness from Comma.Shop it fits perfectly in the 2024 Honda Accord hybrid. Now it seems that the original Comma software doesn’t support the 2024 accord. Anyone know if SunnyPilot, DragonPilot or another fork will work?

Kevmaninc, 2025-05-04:
Yes, go to Discord and join the community there are lots of forks that work
with that car.