commaai/opendbc #1299: GM SDGM Harness

garrettpall, 2024-09-27:
- [X] which car is this for: 2023 Cadillac XT4/XT5/XT6, 2023 Chevy Traverse, 2023 Buick Baby Enclave (CN), 2019 Chevy Volt, 2019 Buick Enclave 
- [ ] schematic: https://github.com/garrettpall/GMSDGMConnector
- [ ] beta sent to tester
- [ ] in shop
- [ ] Added to openpilot docs

For use with:
https://github.com/commaai/opendbc/pull/1177

KINGBEAKSUU, 2025-01-29:
Saw your SDGM harness is it also applicable to 2020 The New Malibu?

yimidi-kr, 2025-03-05:
I already ordered SDGM harness, It is for Cadillac CT6-2019 (3.6L / Adaptive Cruise Control , no Super Cruise). But I’m not sure SDGM connector is matching. If not, I wish modify it. If you have good information please share with me.

garrettpall, 2025-03-05:
> I already ordered SDGM harness, It is for Cadillac CT6-2019 (3.6L / Adaptive Cruise Control , no Super Cruise). But I’m not sure SDGM connector is matching. If not, I wish modify it. If you have good information please share with me.

Please reach out to me (mochi 23 XT4) on discord