commaai/opendbc #1351: Normal install for GM ASCM cars

adeebshihadeh, 2024-10-05:
Can we make the ASCM GM cars a completely normal install? That means a single harness, and we remove the OBD-II harness and the 3rd party L&P harness.

I haven't looked into this at all yet, so not sure it's possible. If it's not, let's close this.

It's fine to lose functionality, such as longitudinal control. A normal install will get the ASCM cars back on https://comma.ai/vehicles.

In order to claim the bounty:
* post the harness diagram
* must work for the currently supported Chevy Volts
* get a PR merged to move the GM ASCM cars to this new harness

Thinkpad4by3, 2024-11-23:
I have a working port of this running. Harness uses standard car-harness box at the ASCM with a cable running to the front (similar to Nissan Rogue installation). Latest master is running, will follow up with diagram, route and PR tomorrow. Car is a 2018 Chevy Volt Premier w/ ACC.

Thinkpad4by3, 2024-11-23:
Here's what I came up with: 

Harness just intercepts the PT bus at the ASCM. Everything is passed through. 
[image]
[image]

I obtained the connectors here, I am not sure of their actual P/N: https://www.amazon.com/RED-WOLF-Replacement-Connector-Compatible/dp/B099MY1WKR/

Changes were needed in Panda due to quirk with one of the message on GM failing to send after the ASCM boots up. 
OpenDBC PR: https://github.com/commaai/opendbc/pull/1524
Panda PR: https://github.com/commaai/panda/pull/2088

Here is a route of it working on my Volt: 20cb801701fc8237/0000002b--160ea66fef