commaai/opendbc #1388: Hyundai kona 2020 ev no longer detected as supported

andrewchambers, 2024-10-18:
### Describe the bug

My car (Hyundai kona 2020 ev) was detected and working on release 0.9.7, I upgraded to master-ci  and I now get a warning that my car is unsupported and it reverts to dashcam mode.

### Which car does this affect?

Hyundai kona 2020 ev

### Provide a route where the issue occurs

7358a9923f0b36da/00000001--d9a0fbdad6

### openpilot version

4cdc76c

### Additional info

dongle-id=7358a9923f0b36da

[image]
[image]

sshane, 2024-10-18:
Your car was CAN fingerprinting before and was left out of the migration to FPv2. Adding your FW versions here: https://github.com/commaai/opendbc/pull/1387

closed 2024-10-18 by sshane

andrewchambers, 2024-10-18:
Thanks, really appreciate your work.