commaai/opendbc #1395: Honda: Civic Hybrid uses GEARBOX_ALT

jyoung8607, 2024-10-22:
A user with a 2023 Honda Civic Hybrid tried to use openpilot and failed due to the GEARBOX message being missing. The hybrid models appear to use an ["e-CVT"](https://www.driveaccord.net/threads/an-interesting-video-describing-honda-e-cvt-%E2%80%98transmission%E2%80%99.555323/) which isn't really a CVT.

The car does output a GEARBOX_ALT message for which we already have a DBC entry. The owner is up and running with a test change https://github.com/jyoung8607/opendbc/commit/c82e4c971fb17145dcc711e66954e807babc4eb9 that isn't quite ready for upstream. To move forward, auto-detection should be cleaned up and combined with other Honda hybrids, and a separate CarDocs model should be added.

**Preserved route:** `a123803465f6695e/00000004--6c9995af1e`
**Fingerprint diff:** https://github.com/jyoung8607/opendbc/commit/18a44117a7b9c6b90df2e18abf3305e1e38c9e84 (we don't get transmission FW with the current query)

Gigafrost, 2024-12-16:
Any updates on this? I own a 2025 Civic hybrid hatchback and I think I have a similar issue.

jyoung8607, 2024-12-18:
> Any updates on this? I own a 2025 Civic hybrid hatchback and I think I have a similar issue.

I've sent you a test branch to try on Discord.

vanillagorillaa, 2025-04-13:
Can be closed now https://github.com/commaai/opendbc/commit/f246792d750fc848dc15531dea4580cc8c1efdd5

closed 2025-07-07 by jyoung8607