commaai/opendbc #1422: Several errors in DBC files

Elmue, 2024-10-27:
Hello

I wrote a DBC parser which prints a lot of errors when I feed it with your files:

```
acura_ilx_2016_can_generated.dbc
Error in line 153, column 13: Signal name expected.
Error in line 154, column 13: Signal name expected.
------------------------
acura_rdx_2018_can_generated.dbc
Error in line 153, column 13: Signal name expected.
Error in line 154, column 13: Signal name expected.
------------------------
acura_rdx_2020_can_generated.dbc
Error in line 153, column 13: Signal name expected.
Error in line 154, column 13: Signal name expected.
Error in line 493, column 10: The signal 'ECON_ON_2' has not been declared.
Error in line 496, column 10: The signal 'CMBS_BUTTON' has not been declared.
------------------------
chrysler_cusw.dbc
Error in line 186, column 23: The message 1492 (0x5D4) has not been declared.
Error in line 187, column 26: The message 498 (0x1F2) has not been declared.
Error in line 188, column 26: The message 498 (0x1F2) has not been declared.
Error in line 190, column 24: The signal 'LKAS_STATE' has not been declared.
------------------------
chrysler_pacifica_2017_hybrid_generated.dbc
Error in line 181, column 1: Semicolon expected.
Error in line 328, column 29: The signal 'BRAKE_PRESSED_2' has not been declared.
Error in line 331, column 25: The signal 'ACCEL_PEDAL' has not been declared.
Error in line 333, column 34: The signal 'LKAS_STEERING_TORQUE' has not been declared.
Error in line 355, column 34: The signal 'ACC_SPEED_CONFIG_KPH' has not been declared.
Error in line 356, column 34: The signal 'ACC_SPEED_CONFIG_MPH' has not been declared.
Error in line 357, column 26: The signal 'CRUISE_STATE' has not been declared.
Error in line 358, column 19: The signal 'SPEED' has not been declared.
Error in line 360, column 10: The signal 'CRUISE_STATE' has not been declared.
------------------------
chrysler_ram_dt_generated.dbc
Error in line 183, column 1: Semicolon expected.
Error in line 186, column 31: The message 258 (0x102) has not been declared.
Error in line 187, column 27: The message 264 (0x108) has not been declared.
Error in line 188, column 36: The message 264 (0x108) has not been declared.
Error in line 189, column 29: The message 678 (0x2A6) has not been declared.
Error in line 190, column 29: The message 678 (0x2A6) has not been declared.
Error in line 191, column 25: The message 678 (0x2A6) has not been declared.
Error in line 244, column 25: The signal 'ACC_Engaged' has not been declared.
Error in line 246, column 28: The signal 'Auto_High_Beam' has not been declared.
Error in line 248, column 36: The signal 'Driver_Seatbelt_Status' has not been declared.
Error in line 249, column 36: The signal 'High_Beam_Lever_Status' has not been declared.
------------------------
chrysler_ram_hd_generated.dbc
Error in line 183, column 1: Semicolon expected.
Error in line 189, column 29: The message 678 (0x2A6) has not been declared.
Error in line 190, column 29: The message 678 (0x2A6) has not been declared.
Error in line 191, column 25: The message 678 (0x2A6) has not been declared.
------------------------
gm_global_a_lowspeed.dbc
Error in line 117, column 16: The signal 'BSM_Indicator_Right' has not been declared.
------------------------
gm_global_a_object.dbc
Error in line 279: Indentifier BO_ not at start of line
Error in line 682: Indentifier BO_ not at start of line
------------------------
gm_global_a_powertrain_generated.dbc
Error in line 359, column 0: Semicolon expected.
------------------------
honda_accord_2018_can_generated.dbc
Error in line 153, column 13: Signal name expected.
Error in line 154, column 13: Signal name expected.
Error in line 520, column 10: The signal 'ECON_ON_2' has not been declared.
Error in line 523, column 10: The signal 'CMBS_BUTTON' has not been declared.
[... 125 lines cut ...]
Error in line 1582, column 28: The signal 'LDW_Direction' has not been declared.
Error in line 1583, column 27: The signal 'XX_DLCORTLC1' has not been declared.
Error in line 1584, column 27: The signal 'XX_DLCORTLC2' has not been declared.
Error in line 1586, column 30: The signal 'MFA_v_Signal_02' has not been declared.
------------------------
vw_mqb_2010.dbc
Error in line 1592, column 30: The signal 'ACC_Hold_Request' has not been declared.
Error in line 1593, column 31: The signal 'ACC_Boost_Request' has not been declared.
Error in line 1594, column 35: The signal 'ACC_Freewheel_Request' has not been declared.
Error in line 1595, column 30: The signal 'ACC_Hold_Release' has not been declared.
Error in line 1596, column 33: The signal 'ACC_Accel_Secondary' has not been declared.
Error in line 1597, column 27: The signal 'ACC_Accel_TSK' has not been declared.
Error in line 1608, column 25: The signal 'LDW_Unknown' has not been declared.
Error in line 1609, column 27: The signal 'Alert_Message' has not been declared.
Error in line 1610, column 27: The signal 'LDW_Direction' has not been declared.
Error in line 1611, column 31: The signal 'Right_Lane_Status' has not been declared.
Error in line 1612, column 30: The signal 'Left_Lane_Status' has not been declared.
Error in line 1613, column 31: The signal 'Kombi_Lamp_Orange' has not been declared.
Error in line 1614, column 30: The signal 'Kombi_Lamp_Green' has not been declared.
Error in line 1615, column 24: The signal 'Folgefahrt' has not been declared.
Error in line 1616, column 24: The signal 'SetAbstand' has not been declared.
Error in line 1617, column 21: The signal 'Abstand' has not been declared.
Error in line 1618, column 22: The signal 'SetSpeed' has not been declared.

```

jyoung8607, 2024-10-27:
This is fair feedback. It's desirable for other DBC tools to be able to open any of these files. Writing a DBC linter has been on our wishlist for a long time, see #579.

In terms of Vector official DBC format compliance, the generated DBCs are the worst offenders, because they're not composed in the right order and use invalid comments. But after that, details like the ones you name should be identified and cleaned up.

Is your tool something you plan to open source? If so, drop us a link. If it's simple, has few external dependencies, and can be adapted for CI use, perhaps @adeebshihadeh could be persuaded to open a bounty. 
We could then start cleaning up these issues and prevent new errors from being merged.

Elmue, 2024-10-29:
Hello  jyoung8607

> Is your tool something you plan to open source?

It is not ready yet. I'am still working on it.
I will implement it into the next version of my software HUD ECU Hacker which is charity-ware:
https://netcult.ch/elmue/hud%20ecu%20hacker

I have been searching for a useful DBC parser on Github but I found nothing that I can use.
Most are written in C or C++ or Python which is useless for me. I need C#.
And mostly the error handling is implemented very poorly.
So I wrote my own parser.

I worked on it the last days and added more functionality. Now it even displays MUCH MORE errors!

I uploaded a beta version for you.
Simply put it into a folder and the DBC files into the same folder.
Then it will parse all files and show the results.

I followed the few rules with sparse descriptions in the PDF file from Vector Informatik GmbH.
However this document is so poor that it is quite useless.
But I implemented nearly all except Environment Variables.
If a section is not checked a hint will be printed like "Section EV_ skipped".
But in your files all sections are parsed.

If you should find any bugs please send me an email, which you see when you start the program.

You can download it here:
https://netcult.ch/elmue/download/DBC_Parser.zip
All you need is .NET framework 4.0 or higher.

And here a screenshot:

[image]

Elmue, 2024-10-30:
I uploaded a new version 1.1 which finds even more errors.

closed 2026-09-15 by adeebshihadeh