commaai/opendbc #1528: Get build times under 1s

adeebshihadeh, 2024-11-26:
I have a modern Threadripper with tons of cores - why is building not instant? Is it the Cython? Is it all the C++ regex?

opendbc is a small project, so working in it should feel very snappy and instant.

It's possible there's one silly thing taking all that time or it's a few things we have to fix. Some notes in #1180.

```
(opendbc) batman@workstation-adeeb:~/src/opendbc$ echo "int main(void) {return 0;}" > test.c; time g++ test.c

real    0m0.041s
user    0m0.016s
sys     0m0.025s
```

---

The goal is to get `time scons` to report <1s (on my workstation, though it shouldn't be too different on other machines). Here's where we are now:
```
(opendbc) batman@workstation-adeeb:~/src/opendbc$ scons -j40
Cannot read termcap database;
using dumb terminal settings.
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
g++ -o opendbc/can/common.os -c -std=c++1z -DDBC_FILE_PATH='"/home/batman/src/opendbc/opendbc/dbc"' -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -fPIC -I. -I/usr/lib/include -I/home/batman/.local/share/uv/python/cpython-3.11.4-linux-x86_64-gnu/include/python3.11 opendbc/can/common.cc
g++ -o opendbc/can/dbc.os -c -std=c++1z -DDBC_FILE_PATH='"/home/batman/src/opendbc/opendbc/dbc"' -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -fPIC -I. -I/usr/lib/include -I/home/batman/.local/share/uv/python/cpython-3.11.4-linux-x86_64-gnu/include/python3.11 opendbc/can/dbc.cc
g++ -o opendbc/can/parser.os -c -std=c++1z -DDBC_FILE_PATH='"/home/batman/src/opendbc/opendbc/dbc"' -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -fPIC -I. -I/usr/lib/include -I/home/batman/.local/share/uv/python/cpython-3.11.4-linux-x86_64-gnu/include/python3.11 opendbc/can/parser.cc
g++ -o opendbc/can/packer.os -c -std=c++1z -DDBC_FILE_PATH='"/home/batman/src/opendbc/opendbc/dbc"' -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -fPIC -I. -I/usr/lib/include -I/home/batman/.local/share/uv/python/cpython-3.11.4-linux-x86_64-gnu/include/python3.11 opendbc/can/packer.cc
cythonize opendbc/can/packer_pyx.pyx
cythonize opendbc/can/parser_pyx.pyx
python3 opendbc/dbc/generator/generator.py
Compiling /home/batman/src/opendbc/opendbc/can/packer_pyx.pyx because it changed.
[1/1] Cythonizing /home/batman/src/opendbc/opendbc/can/packer_pyx.pyx
Compiling /home/batman/src/opendbc/opendbc/can/parser_pyx.pyx because it changed.
[1/1] Cythonizing /home/batman/src/opendbc/opendbc/can/parser_pyx.pyx
g++ -o opendbc/can/packer_pyx.o -c -std=c++1z -g -fPIC -O2 -Wunused -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I. -I/usr/lib/include -I/home/batman/.local/share/uv/python/cpython-3.11.4-linux-x86_64-gnu/include/python3.11 -I.venv/lib/python3.11/site-packages/numpy/_core/include opendbc/can/packer_pyx.cpp
g++ -o opendbc/can/parser_pyx.o -c -std=c++1z -g -fPIC -O2 -Wunused -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I. -I/usr/lib/include -I/home/batman/.local/share/uv/python/cpython-3.11.4-linux-x86_64-gnu/include/python3.11 -I.venv/lib/python3.11/site-packages/numpy/_core/include opendbc/can/parser_pyx.cpp
g++ -o opendbc/can/libdbc.so -shared opendbc/can/dbc.os opendbc/can/parser.os opendbc/can/packer.os opendbc/can/common.os -Lopendbc/can
g++ -o opendbc/can/parser_pyx.so -pthread -shared -Wl,-rpath=/home/batman/src/opendbc/opendbc/can opendbc/can/parser_pyx.o -Lopendbc/can -Lopendbc/can -ldbc
g++ -o opendbc/can/packer_pyx.so -pthread -shared -Wl,-rpath=/home/batman/src/opendbc/opendbc/can opendbc/can/packer_pyx.o -Lopendbc/can -Lopendbc/can -ldbc
scons: done building targets.
```

LouisVelez, 2024-11-26:
It seems most of the delay is on the build. I would suggest looking into pre-compiling if possible. Hopefully this helps anyone else looking into the issue. 

Output of  `scons -u -j$(nproc) --debug=time`  on an M2 mac

```
Total SConsign sync time: 0.002967 seconds
Total build time: 2.888204 seconds
Total SConscript file execution time: 0.216717 seconds
Total SCons execution time: 0.050112 seconds
Total command execution time: 2.621375 seconds
```

qedjdev, 2024-12-05:
> It seems most of the delay is on the build. I would suggest looking into pre-compiling if possible

target: a _clean_ build (scons -c && ...) being <1s
precompiled headers still need to be compiled on a clean build. this doesn't save time _for this target_ but it should still (maybe?) be looked into.

for anyone willing to dig through profiling data...

`scons -Q --profile build_profile.pstats`

```python
import pstats
p = pstats.Stats('build_profile.pstats')
p.sort_stats('cumulative').print_stats()
```

Riley-livingston, 2024-12-15:
@adeebshihadeh What is your current build time benchmark in seconds on your machine when running `scons -c && scons -u -j40  --debug=time` ?

 I replaced the C++ regex with [RE2](https://github.com/google/re2) and it sped up the build times from clean build quite significantly on my low-end Mac.

Arjun31415, 2024-12-27:
I used ClangBuildAnalyzer and this is what I get - 
```
Analyzing build trace from 'clang_build_analysis.txt'...
**** Time summary:
Compilation (12 times):
  Parsing (frontend):            5.9 s
  Codegen & opts (backend):      8.3 s

**** Files that took longest to parse (compiler frontend):
  1571 ms: opendbc//can/dbc.json
   952 ms: opendbc//can/parser_pyx.o
   756 ms: opendbc//can/packer_pyx.o
   659 ms: opendbc//can/packer.json
   657 ms: opendbc//can/parser.json
   463 ms: opendbc//can/dbc.json
   408 ms: opendbc//can/common.json
   170 ms: opendbc//can/parser_pyx.o
    85 ms: opendbc//can/packer_pyx.o
    82 ms: opendbc//can/parser.json

**** Files that took longest to codegen (compiler backend):
  3773 ms: opendbc//can/dbc.json
  2587 ms: opendbc//can/parser_pyx.o
   980 ms: opendbc//can/packer_pyx.o
   517 ms: opendbc//can/parser.json
   269 ms: opendbc//can/packer.json
   168 ms: opendbc//can/common.json

**** Templates that took longest to instantiate:
   338 ms: std::basic_regex<char>::basic_regex (1 times, avg 338 ms)
   338 ms: std::basic_regex<char>::_M_compile (1 times, avg 338 ms)
   327 ms: std::__detail::_Compiler<std::regex_traits<char>>::_Compiler (1 times, avg 327 ms)
   283 ms: std::__detail::_Compiler<std::regex_traits<char>>::_M_disjunction (1 times, avg 283 ms)
   281 ms: std::__detail::_Compiler<std::regex_traits<char>>::_M_alternative (1 times, avg 281 ms)
   277 ms: std::__detail::_Compiler<std::regex_traits<char>>::_M_term (1 times, avg 277 ms)
   236 ms: std::__detail::_Compiler<std::regex_traits<char>>::_M_atom (1 times, avg 236 ms)
    84 ms: std::__detail::_Compiler<std::regex_traits<char>>::_M_insert_charact... (1 times, avg 84 ms)
    71 ms: std::regex_match<std::char_traits<char>, std::allocator<char>, std::... (1 times, avg 71 ms)
    71 ms: std::regex_match<__gnu_cxx::__normal_iterator<const char *, std::bas... (1 times, avg 71 ms)
    71 ms: std::__detail::__regex_algo_impl<__gnu_cxx::__normal_iterator<const ... (1 times, avg 71 ms)
    53 ms: std::unordered_map<int, int> (6 times, avg 8 ms)
    44 ms: std::unordered_map<unsigned int, const Msg *> (6 times, avg 7 ms)
    43 ms: std::unordered_map<std::basic_string<char>, const Msg *> (6 times, avg 7 ms)
    42 ms: std::_Hashtable<int, std::pair<const int, int>, std::allocator<std::... (6 times, avg 7 ms)
    39 ms: std::__detail::_Compiler<std::regex_traits<char>>::_M_quantifier (1 times, avg 39 ms)
    37 ms: std::_Hashtable<std::basic_string<char>, std::pair<const std::basic_... (6 times, avg 6 ms)
    36 ms: std::__detail::_Executor<__gnu_cxx::__normal_iterator<const char *, ... (1 times, avg 36 ms)
    36 ms: std::__detail::_Executor<__gnu_cxx::__normal_iterator<const char *, ... (1 times, avg 36 ms)
    36 ms: std::__detail::_Executor<__gnu_cxx::__normal_iterator<const char *, ... (1 times, avg 36 ms)
    35 ms: std::unordered_map<unsigned int, MessageState> (6 times, avg 5 ms)
    35 ms: std::_Hashtable<unsigned int, std::pair<const unsigned int, const Ms... (6 times, avg 5 ms)
    31 ms: std::__detail::_Compiler<std::regex_traits<char>>::_M_bracket_expres... (1 times, avg 31 ms)
    30 ms: std::_Hashtable<unsigned int, std::pair<const unsigned int, MessageS... (6 times, avg 5 ms)
    28 ms: std::__detail::_StateSeq<std::regex_traits<char>>::_M_clone (1 times, avg 28 ms)
    27 ms: std::vector<Signal>::operator= (3 times, avg 9 ms)
    27 ms: std::__detail::_BracketMatcher<std::regex_traits<char>, false, false... (1 times, avg 27 ms)
    25 ms: std::unordered_map<unsigned int, MessageState>::at (6 times, avg 4 ms)
    25 ms: std::__detail::_Map_base<unsigned int, std::pair<const unsigned int,... (6 times, avg 4 ms)
    24 ms: std::map<std::pair<unsigned int, std::basic_string<char>>, Signal>::... (1 times, avg 24 ms)

**** Template sets that took longest to instantiate:
   338 ms: std::basic_regex<$>::basic_regex (1 times, avg 338 ms)
   338 ms: std::basic_regex<$>::_M_compile (1 times, avg 338 ms)
   327 ms: std::__detail::_Compiler<$>::_Compiler (1 times, avg 327 ms)
   283 ms: std::__detail::_Compiler<$>::_M_disjunction (1 times, avg 283 ms)
   281 ms: std::__detail::_Compiler<$>::_M_alternative (1 times, avg 281 ms)
   277 ms: std::__detail::_Compiler<$>::_M_term (1 times, avg 277 ms)
   236 ms: std::__detail::_Compiler<$>::_M_atom (1 times, avg 236 ms)
   182 ms: std::unordered_map<$> (25 times, avg 7 ms)
   164 ms: std::_Hashtable<$> (31 times, avg 5 ms)
[... 111 lines cut ...]
  1x: filesystem fs_fwd.h system_error 

296 ms: /usr/include/c++/14.2.1/algorithm (included 3 times, avg 98 ms), included via:
  3x: <direct include>

273 ms: /usr/include/c++/14.2.1/bits/ios_base.h (included 4 times, avg 68 ms), included via:
  2x: <direct include>
  1x: sstream istream ios 
  1x: filesystem fs_path.h locale locale_facets.h 

192 ms: opendbc/can/common_dbc.h (included 6 times, avg 32 ms), included via:
  3x: common.h 
  2x: <direct include>
  1x: cstring 

160 ms: /usr/include/c++/14.2.1/cassert (included 2 times, avg 80 ms), included via:
  2x: <direct include>

  done in 0.0s.
  ```

The `.json` files can be ignored because they are file generated by `-ftime-trace`

Regex takes the longest to instantiate. 
`opendbc/can.common.h` is being included multiple(6 times) causing significant slowdown (500ms)

Riley-livingston, 2024-12-27:
> I used ClangBuildAnalyzer and this is what I get -
>
> ```
>
> Analyzing build trace from 'clang_build_analysis.txt'...
>
> **** Time summary:
>
> Compilation (12 times):
>
>   Parsing (frontend):            5.9 [...]

So it looks like the most impactful change would be optimizing or replacing regex with a lighter implementation(~25% of the compilation bottleneck in clean builds).

Also, what if we split `common.h` into smaller, focused headers to reduce the compilation unit size? Might also be a good idea to review circular dependencies between common.h and common_dbc.h

MikeBusuttil, 2024-12-30:
@adeebshihadeh I believe I've got this one solved: https://github.com/commaai/opendbc/pull/1602

MikeBusuttil, 2025-01-13:
I dug into this 1 over the past little bit and go back to work tomorrow so I wanted to share my findings in case anyone wants to move forward on this while I'm preoccupied.

Running `scons -c && scons -j$(nproc)` results in 12 commands (detailed below) being run after (and while) scons does some work to determine what those 12 commands will be.  Here's a graphical representation of that:
```mermaid
graph TB;
    classDef bottleneck stroke:#f00
    A[g++ -o common.os<br>.3 s<br>];
    B[g++ -o dbc.os<br>1.0 s<br><br><br><br><br><br><br><br>];
    C[g++ -o parser.os<br>.6 s<br><br><br><br>];
    D[g++ -o packer.os<br>.5 s<br><br><br>];
    E[g++ -o libdbc.so - .1 s];
    F[cythonize packer_pyx.pyx<br>.5 s<br><br><br>];
    G[g++ -o packer_pyx.o<br>.7 s<br><br><br><br><br>];
    H[g++ -o packer_pyx.so - .1 s];
    I[cythonize parser_pyx.pyx<br>.6 s<br><br><br><br>]:::bottleneck;
    J[g++ -o parser_pyx.o<br>1.1 s<br><br><br><br><br><br><br><br><br>]:::bottleneck;
    K[g++ -o parser_pyx.so - .1 s]:::bottleneck;
    L[python3 generator.py];
    Scons[Scons overhead<br>.9 s<br><br><br><br><br><br><br>]:::bottleneck;
    A-->E;
    B-->E;
    B-->H;
    B-->K;
    C-->E;
    D-->E;
    F-->G;
    G-->H;
    I-->J;
    J-->K;
    Scons-->A;
    Scons-->B;
    Scons-->C;
    Scons-->D;
    Scons-->F;
    Scons-->I;
    Scons-->L;
    Start-->Scons;
```
- The size of the job is related to the duration of the job.  Times (on my machine) are listed inside each job.
- The jobs highlighted with red are the bottleneck.  Improving the speed on any of those should improve the resulting build time.

## Increasing speed of `g++ -o dbc.os`

On `master`, compiling `dbc.cc` is the bottleneck.  It uses the c++ standard regex library which has very poor build-time performance.  Replacing this with [RE2](https://github.com/google/re2) (done in #1602) yields a 5-10x performance increase for that job at the cost of adding an additional dependency.

## Increasing the speed of `g++ -o parser_pyx.o`

### compiling without optimization
This is another significant bottleneck on `master` that was addressed in #1602 by compiling without optimization (using `-O0` instead of `-O2`).  Making this change yields a 5-10x performance increase for that job at the cost of a less-optimized run-time during development.

### splitting into multiple steps
To further increase the speed of this step it can be broken into 2 files that can be `cythonize`d and compiled in parallel for modest performance gains (~10% for that job).  The gains are minimal and it comes at the cost of a little complexity and resource inefficiency.  This work was done [here](https://github.com/MikeBusuttil/opendbc/pull/2/files) and can be merged if the trade-off's are deemed worth it.

## Increasing the speed of `cythonize parser_pyx.pyx` + `g++ -o parser_pyx.o` steps

There's not much that can be done here without moving away from cython.  Using cython means having about 0.8s of overhead (see "Cython overhead" below) before you can do anything useful.  To improve these step any further would require a big refactor like an architecture change or rewriting these in C++ since Python is likely out of the question for runtime performance.

## Increasing the speed of the `scons` overhead

I have not exhausted all the options here and believe there is still some fat yet to be trimmed.  For example, saving the path which numpy stores its header files on your local machine will save you ~15% on overall build performance but this comes at the cost of code complexity (see [here](https://github.com/MikeBusuttil/opendbc/pull/3/files#diff-d78c5c377c03dd251edc72cdb0ae7d9986fa08361eb392187218f8c670c0bc4e) for what that could look like).  To see what kind of a dent you're making, run `scons` with the `--debug=time` flag.

Another option is to cache all the build steps and bypass scons entirely.  This gets the build times well under 1s (see [here](https://github.com/MikeBusuttil/opendbc/pull/3/files#diff-18316ed146f6e7275ad1467df4f71259bd15df1b9adc6b74c92c33649a65362f) for a node script that does that) but is pretty hacky and presumably not the direction we want to go in.

## Good luck!

If you made it this far and are motivated to proceed I'd be down to pair on it - book some time with me [here](https://calendly.com/mikebusuttil) and let's get 'er done.  Otherwise I may return to it when I get some availability or something comes to me while my mind wanders 🛌🚽🚿

## Addendum

### Cython overhead
[... 29 lines cut ...]
Compiling /root/opendbc/opendbc/can/parser_pyx.pyx because it changed.
[1/1] Cythonizing /root/opendbc/opendbc/can/parser_pyx.pyx

real    0m0.428s
user    0m0.368s
sys     0m0.038s

real    0m0.366s
user    0m0.320s
sys     0m0.027s
```

### Commands resulting from `scons`
1. `g++ -o opendbc/can/common.os -c -std=c++1z -DDBC_FILE_PATH='"/root/opendbc/opendbc/dbc"' -g -fPIC -O0 -Wunused -Werror -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -fPIC -I. -I/usr/lib/include -I/opt/homebrew/include -I/usr/include/python3.12 opendbc/can/common.cc`
2. `g++ -o opendbc/can/dbc.os -c -std=c++1z -DDBC_FILE_PATH='"/root/opendbc/opendbc/dbc"' -g -fPIC -O0 -Wunused -Werror -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -fPIC -I. -I/usr/lib/include -I/opt/homebrew/include -I/usr/include/python3.12 opendbc/can/dbc.cc`
[... cut ...]
10. `g++ -o opendbc/can/libdbc.so -shared opendbc/can/dbc.os opendbc/can/parser.os opendbc/can/packer.os opendbc/can/common.os -Lopendbc/can -L/opt/homebrew/lib`
11. `g++ -o opendbc/can/packer_pyx.so -pthread -shared -Wl,-rpath=/root/opendbc/opendbc/can opendbc/can/packer_pyx.o -Lopendbc/can -L/opt/homebrew/lib -Lopendbc/can -ldbc`
12. `g++ -o opendbc/can/parser_pyx.so -pthread -shared -Wl,-rpath=/root/opendbc/opendbc/can opendbc/can/parser_pyx.o -Lopendbc/can -L/opt/homebrew/lib -Lopendbc/can -ldbc -lre2`

coder351, 2025-07-09:
@adeebshihadeh I have had some success in improving build time but wanted to hold off on raising PR before getting confirmation.

Summary:

1) Replace scons with cmake and ninja. Scons runs much slower than cmake, benchmark attached for reference

| Scons                        | Cmake                                                  |
| -----------------------------| ------------------------------------------------------ |
|scons: done building targets. |ninja: Entering directory `build'                       |
|real    0m8.309s              |real    0m1.550s                                        |
|user    0m27.472s             |user    0m3.859s                                        |
|sys     0m1.675s              |sys     0m0.427s                                        |

2) Use LLVM::Regex instead of std::regex in dbc.cc when parsing dbc from stream.

3) Use precompiled headers, cmake can generate precompile headers during configuration step.

4) Remove cython *.pyx *.pxd with ctypes.Cythonizing step was causing slow down along with the use of std:regex in dbc.cc

5) Use lld for linker instead of ld. lld is linker from llvm and runs much faster than ld.

| lld                                                  |  ld(default linker)                                    |
| -----------------------------------------------------| -------------------------------------------------------|
|ninja: Entering directory `build'                     |ninja: Entering directory `build'                       |
|real    0m1.550s                                      |real    0m2.654s                                        |
|user    0m3.859s                                      |user    0m5.856s                                        |
|sys     0m0.427s                                      |sys     0m0.589s                                        |

Please try out the test [branch](https://github.com/coder351/opendbc/tree/improve-build). To test, checkout improve-build and run,
```
// installs llvm and lld
source ./setup.sh
// configures opendbc using ninja and generates pch files
cmake -G Ninja -S . -B build
// run ninja, replace 8 with # of procs available
time ninja -C build -j8
```

greylot, 2025-07-12:
This SConstruct file can optimize the build process.
```
env = Environment(CCFLAGS=['-std=c++1z', '-O3', '-fPIC'],
                  CPPPATH=['.', '/usr/lib/include', '/path/to/python/include'],
                  LIBPATH=['opendbc/can'])
env.SharedLibrary('libdbc', ['common.cc', 'dbc.cc', 'parser.cc', 'packer.cc'])
env.Cython('packer_pyx.pyx', CYTHONFLAGS=['-j40'])
env.Cython('parser_pyx.pyx', CYTHONFLAGS=['-j40'])
env.Command('generated_files', 'dbc/generator/generator.py', 'python3 $SOURCE')
```

Note:If a specific step, such as generator.py or regex, is taking too much time, further optimization can be focused on that step.

coder351, 2025-07-16:
@sshane / @robbederks I noticed issue commaai/panda/issues/2171 which is about using source + header file structure. The opendbc repo also has libsafety, having _declarations.h files. The starting point for the panda issue for using source + header file structure can be the opendbc safety folder. I also noticed pullrequest commaai/panda/pull/2212/files on line 39, the suggestion from codex is to use

faults = env.SharedObject("faults.os", File("#/board/faults.c"))

I tried to follow the suggestion but in scons under board e.g stm32f4 and stm32h7 have multiple versions of llfan.h which would split in llfan.c and llfan.h. Scons is not able to differentiate llfan.c based on parent folder i.e stm32f4 or stm32h7 and throws error.
I suggest trying out using cmake in the current issue and also split safety folder to use source + header file structure. Then if the solution proves effective, refactor panda to use cmake instead of scons. CMakeLists are easy to learn and implement compared to scons which is difficult to maintain. 

If no objections, I am able to raise PR for opendbc which addresses the current issue of build times along with safety folder to use source + header file structure using cmake and if cmakelist prove to be useful, then also migrate panda to cmake.

PR #2539 (2025-07-23): CANParser: python rewrite
Resolves #1528
closed 2025-07-23 by adeebshihadeh