commaai/opendbc #1572: Santa fe 2019

thabet33, 2024-12-12:
### Describe the bug

Dashcam mode car unrecognized
Please help
d7337b1b1b253a32/00000000--84c4329f82

### Which car does this affect?

Hyundai santafe 

### Provide a route where the issue occurs

d7337b1b1b253a32/00000000--84c4329f82

### openpilot version

0.9.7

sshane, 2024-12-12:
It looks like your car does not have the required features needed to interface with your car; Smart Cruise Control (SCC). I see no radar FW version in your logs. If this is incorrect, please comment back.

closed 2024-12-12 by sshane

thabet33, 2024-12-13:
thats correct 
dose it mean it will not work ? 
because i see all santa fe is suppuorted in the cars list

sshane, 2024-12-13:
Yes unfortunately, it will not work. The docs aren't fully accurate for global models as we aim to target the US market. In the US all trims come with SCC so that's why we say it will work with all variants, but your region it seems does not come with SCC on your trim.

thabet33, 2024-12-13:
Thank you for your time