commaai/opendbc #1574: Car Unrecognized: Honda Civic Hybrid Hatchback

Gigafrost, 2024-12-14:
### Describe the bug

I'm getting the following errors when running stock openai or the master-ci.

Dashcam mode: Car not recognized and CAN Error: Check Connection.

### Which car does this affect?

2025 Honda Civic Hatchback Hybrid Sport touring 

### Provide a route where the issue occurs

4d4cdf94242eb4cf/00000002--b5901de348

### openpilot version

0.9.7

### Additional info

This happened on both the stock version of open pilot and on the master-ci version. I tried the nightly version of open pilot as well which also gave the same issues.

sshane, 2024-12-15:
The logs indicate that the gearbox message we rely on is missing, you'll need to find it for your car. You can switch into different gears and use [Cabana](https://github.com/commaai/openpilot/tree/master/tools/cabana) to explore the CAN data to find the new gear signal.

Gigafrost, 2024-12-15:
Gotcha, I never used Cabana before. Is there. A tutorial I can follow that teaches me how to use it?

sshane, 2024-12-15:
Here's some resources:

https://blog.comma.ai/how-to-write-a-car-port-for-openpilot
https://www.youtube.com/watch?v=KcfzEHB6ms4

To see if you can quickly find it, check our DBCs for other Honda `GEARBOX` addresses to see if they exist

Gigafrost, 2024-12-16:
I'm still having some issues finding it. Would it be possible for me to just send you my data instead to make a fork from? As I mentioned I am terrible at coding and don't seem to be getting very far in terms of knowing how to do what.

jyoung8607, 2024-12-18:
Confirmed duplicate of #1395.

closed 2024-12-18 by jyoung8607