commaai/opendbc #1581: Python 3.11.7 failed to build the opendbc

helperfunc, 2024-12-15:
```
>scons -j8
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
g++ -o opendbc\can\dbc.os -c -std=c++1z -DDBC_FILE_PATH='"C:\Users\huixu3\code\opendbc\opendbc\dbc"' -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -I. -IC:\usr\lib\include -IC:\Users\huixu3\AppData\Local\Programs\Python\Python311\Include opendbc\can\dbc.cc
g++ -o opendbc\can\packer_pyx.o -c -std=c++1z -g -fPIC -O2 -Wunused -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I. -IC:\usr\lib\include -IC:\Users\huixu3\AppData\Local\Programs\Python\Python311\Include -IC:\Users\huixu3\venv\Lib\site-packages\numpy\_core\include opendbc\can\packer_pyx.cpp
g++ -o opendbc\can\parser_pyx.o -c -std=c++1z -g -fPIC -O2 -Wunused -Wshadow -Wno-vla-cxx-extension -Wno-unknown-warning-option -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I. -IC:\usr\lib\include -IC:\Users\huixu3\AppData\Local\Programs\Python\Python311\Include -IC:\Users\huixu3\venv\Lib\site-packages\numpy\_core\include opendbc\can\parser_pyx.cpp
python3 opendbc\dbc\generator\generator.py
Traceback (most recent call last):
  File "C:\Users\huixu3\code\opendbc\opendbc\dbc\generator\generator.py", line 64, in <module>
    create_all(opendbc_root)
  File "C:\Users\huixu3\code\opendbc\opendbc\dbc\generator\generator.py", line 49, in create_all
    subprocess.check_call(f)
  File "C:\Users\huixu3\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 408, in check_call
    retcode = call(*popenargs, **kwargs)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\huixu3\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 389, in call
    with Popen(*popenargs, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\huixu3\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\huixu3\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 1538, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [WinError 193] %1 is not a valid Win32 application
scons: *** [opendbc\dbc\chrysler_ram_dt_generated.dbc] Error 1
opendbc\can\packer_pyx.cpp:1277:10: fatal error: common_dbc.h: No such file or directory
 1277 | #include "common_dbc.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
scons: *** [opendbc\can\packer_pyx.o] Error 1
opendbc\can\parser_pyx.cpp:1277:10: fatal error: common_dbc.h: No such file or directory
 1277 | #include "common_dbc.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
scons: *** [opendbc\can\parser_pyx.o] Error 1
opendbc\can\dbc.cc: In function 'const std::string get_dbc_root_path()':
<command-line>: error: incomplete universal character name \U
<command-line>: note: in definition of macro 'DBC_FILE_PATH'
<command-line>: error: unknown escape sequence: '\h' [-Werror]
<command-line>: note: in definition of macro 'DBC_FILE_PATH'
<command-line>: error: unknown escape sequence: '\c' [-Werror]
<command-line>: note: in definition of macro 'DBC_FILE_PATH'
<command-line>: error: unknown escape sequence: '\o' [-Werror]
<command-line>: note: in definition of macro 'DBC_FILE_PATH'
<command-line>: error: unknown escape sequence: '\o' [-Werror]
<command-line>: note: in definition of macro 'DBC_FILE_PATH'
<command-line>: error: unknown escape sequence: '\d' [-Werror]
<command-line>: note: in definition of macro 'DBC_FILE_PATH'
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
cc1plus: note: unrecognized command-line option '-Wno-vla-cxx-extension' may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
scons: *** [opendbc\can\dbc.os] Error 1
scons: building terminated because of errors.
```

adeebshihadeh, 2024-12-15:
Need way more info. Also try `./test.sh`.

closed 2024-12-15 by adeebshihadeh

jyoung8607, 2024-12-15:
At a glance, it looks like you're trying to build on Windows natively. openpilot and opendbc are designed and supported to run on Ubuntu 24.04.