commaai/opendbc #1586: Cruise Fault when powering the car for the 1st time in the day

renatosc, 2024-12-17:
### Describe the bug

I have been using comma 3 for 1.5 years now and in the last few months I started to get the "Cruise fault" error a lot (I don't think I ever had that error on the first year using Comma). 

I was able to identify that the error happens when I am powering on my car usually in the morning and the Comma was totally powered off. 

So my car powers on while the Comma is still loading and then Comma indicates that that Car is offline and only after Comma indicates the Car is online I can then power off and on my car and the error goes away (If I try to power off/on the car while Comma indicates the Car is offline, it will still throw the Cruise fault error).

### Which car does this affect?

Honda Oyssey 2019

### Provide a route where the issue occurs

n/a

### openpilot version

0.9.7

renatosc, 2025-01-04:
@sshane sorry for the delay. I haven't seen the issue again since I opened this issue until yesterday. What do you need to me to send to you about yesterday's episode? (what you mean by "route"?)

sshane, 2025-01-13:
Find where it happens and send the link from https://connect.comma.ai/

renatosc, 2025-01-13:
I will do that, thank you (Last week route is not appearing anymore in connect.comma.ai)

renatosc, 2025-01-16:
@sshane it happened yesterday and I went to connect.comma.ai and cannot see the the recordings there. I see all recordings before and after the episode, but not the one with the Cruise fault. Maybe Comma does not record when it has Cruise fault?

sshane, 2025-01-30:
Let me know when this happens again and post your dongle id.

renatosc, 2025-01-30:
@sshane it happened again yesterday, and again no recording is available in connect.comma.ai. I have recordings before and after the event. At this point I am pretty sure that then that cruise fault happens, no recording is made.

sshane, 2025-01-30:
We still need your dongle id or nothing can be looked into

renatosc, 2025-01-30:
If the dongle id is the one I see on the url (and below of "comma 3" label on the page), then it is **7a7eba6f9ce36254**

renatosc, 2025-01-30:
Last episode was yesterday, Jan 29, at 7:55am CT

renatosc, 2025-04-14:
This has come back with full force now. It is happening every day, sometimes even twice a day.

At least now I am able to always reproduce it: with the car running and the device working, just do a Device Reboot, after it reboots it will be showing "NO PANDA" for a while and then my car starts throwing all the warnings. Then ~40 seconds later device shows "Vehicle ONLINE" and shows the "Cruise Fault: Restart the car to engage". After restarting the car all the car warnings go away and the device works fine.

mvl-boston, 2025-05-06:
Can you look for your route in connect.aks.comma.ai and post?   It can see more logs than connect.comma.ai

However, good chance you have a worn out USB cable.   Those cause brake faults which cause logs to not fully generate, so matches the symptoms.

renatosc, 2025-05-06:
Thanks for the reply, indeed I can see the route using connect.asks.comma.ai but not with connect.comma.ai.

This is the route: https://connect.aks.comma.ai/7a7eba6f9ce36254/0000099d--dc88c26457

The title of route shows "Invalid date" and below it shows "Invalid date to Invalid date"

[image]

I will replace the usb cable and report back but I don't it is the cable since this only happens during first (cold) boot of comma. It is like comma is taking too long to finish the boot and thus the car times out the waiting and shows all the warnings. If it was the cable, it should show the error very randomly, shouldn't it?

mvl-boston, 2025-05-06:
I'm a community contributor, not a comma employee.  If you want to make the log public, I can dig into whether it is behaving similar to other bad USB wiring problems I've seen.

renatosc, 2025-05-06:
made it public now

mvl-boston, 2025-05-07:
My bet is on a bad USB-C cable.

The car is complaining about an ACC problem even before comma starts logging.  Usually that means the car computer isn't getting all the signals it wants from somewhere, and if it used to work for months and then stopped, and if no one else is complaining, then most likely you have a wiring issue.

From posts online it seems the USB-C cable is the first wire to go bad with age, so worth a try first before more complex debugging.  I could see your first-drive symptom being a cold vs hot wire thing, where the wire warms up and expands and works, and a cold wire doesn't.

What I see in Comma discord is that any USB C cable meeting USB 3.1 Gen II or Thunderbolt III specs or better would work, and solves the problem most of the time.

renatosc, 2025-05-07:
Thank you for taking the time to check the logs. I had read in discord in the past about USB cables going bad and I even bought an extra pair at that time. So I just went ahead to replace the cable and I see it does not fit... So for now I had replaced with another usb-c to usb-a cable that I had here which of course requires me to use a usb-a to usb-c adapter as well. After replacing I saw that comma detected my vehicle very fast. So, first impression is positive. I will run with that new cable for a few days (and also order a better usb-c to usb-c cable) and update here.

Thanks again @mvl-boston

sshane, 2025-05-07:
Sounds similar to https://github.com/commaai/opendbc/issues/2166

Do you know how to ssh in? You can disable panda flashing in pandad by just returning panda here and see if the fault goes away: https://github.com/commaai/openpilot/blob/aa1b7907086b5c1fe5e7b07c92e5aa6090a40c23/selfdrive/pandad/pandad.py#L26

renatosc, 2025-05-11:
After few days of testing with 3 different cables (2 being brand new ones), the problem still happens. I can ssh and change code. So just `return panda` after line 26, right?

Yes, I hear clicking sounds coming from it exactly at the time that I get the erros from the vehicle.

renatosc, 2025-05-14:
The code change did not make any difference, still seeing the error during first boot.

[image]

My next tentatives will be:
1) Try another harness
2) Downgrade Openpilot version
3) Use an openpilot fork (such as SunnyPilot)

sshane, 2025-05-14:
@renatosc can you try this branch?

https://github.com/commaai/openpilot/tree/cruise-faults

renatosc, 2025-05-14:
I can try that @sshane , but I just replaced my harness with an old that I have from a Comma 2 and so far (already did 2 full boots) and got no errors. I will keep driving with the old hardness for a couple days and update this thread.

renatosc, 2025-05-15:
I talked too soon, the error is back with the other harness as well. I will try the [cruise-faults](https://github.com/commaai/openpilot/tree/cruise-faults) branch.

renatosc, 2025-05-19:
@sshane after few days running and rebooting it multiple times, happy to report that I got zero cruise faults errors using your custom branch. Thank you for creating that fix.