commaai/opendbc #1625: COMMA_HACK TODOs

adeebshihadeh, 2025-01-12:
A list of things that should probably get done before the [hackathon](https://blog.comma.ai/comma-hack-5/):
* move panda safety here
* update [commaCarSegments](https://huggingface.co/datasets/commaai/commaCarSegments)

sshane, 2025-01-27:
We're looking into getting these cars:

- Rivian
- F-150 or Mache-E
- Tesla Model Y (ACQUIRED)
- @jyoung8607's pick (ID4?)
- @sunnyhaibin's pick (EV9, x, x, x)
- Subaru angle car?
- New Honda?

marshallfarthing, 2025-02-01:
@sshane I have a 2025 Outback, willing to take a look for a Subaru Angle car?

sshane, 2025-02-04:
Are you in the San Diego area? We are going to rent the Forester 2023 w/ angle control, as the Forester wasn't available. Our docs only show Forester support up to 2023, but the 2025 could likely be supported, it just hasn't been checked. @jnewb1 do you know anything about this?

marshallfarthing, 2025-02-04:
> angle control, as the Forester wasn't available. Our docs only show Outback

I am located in San Francisco, but happy to assist with some dev work! I
have been talking to a few people on discord, and some are claiming the
2024 and 2025 models have the Toyota security enabled, but I don't think
that's true (this isn't a TNGA car and not shared with Toyota)

I attempted fingerprinting on the MY 2025 and it failed, investigating
further with Cabana and it appears that a lot of the CAN messaging has
changed between 2023 and 2025 (I concluded many of the Cruise messages have
moved around), I unfortunately am not super well versed in CAN so I'm
personally at my limits of expertise on it. Happy to loan my car to help
further.

hiimisaac, 2025-02-11:
I noticed that the F150 was removed from Comma Hack testing. With the validation for the same platform being done on the Mach E (Q4 Ford), this will still remove the F150 from dashcam or no?

kjanoudi, 2025-02-13:
commenting to follow

closed 2025-03-03 by adeebshihadeh