commaai/opendbc #1629: Wrong minimum speed to engage on 2018 Acura RDX

rinterdonato, 2025-01-13:
### Describe the bug

My 2018 Acura RDX used to engage OpenPilot at 22mph. On the latest version of the code, it only engages at 25mph. I spoke with Shane about this issue at CES 2025 and he recommended reporting thr behavior as a bug.

### Which car does this affect?

Acura RDX 2018

### Provide a route where the issue occurs

All routes

### openpilot version

0.9.7

### Additional info

The Acura RDX that exhibits this behavior is also equipped with a (now non-functional) Comma Pedal. If the Comma Pedal is the cause of the higher engagement speed, then I can take it out of the car. I have been keeping it in because I have been hoping to get stop&go back in a future version, or at least the ability to engage OpenPilot when stopped behind a lead car again.

sshane, 2025-01-13:
It looks like the pedal was allowing your car to enable down to 0, but maybe your car prevented enabling at 22 mph: https://github.com/commaai/openpilot/blob/0eba392cf24e5ebbd09d7a3ff4acd9f98daeb3bb/selfdrive/car/honda/interface.py#L299-L300

Can you change [this line](https://github.com/commaai/opendbc/blob/2ecc45c4eda285aa0135624a890fd3a17146cae5/opendbc/car/honda/interface.py#L205) to -1 and see:
1. when your car allows you to enable
2. when your car is able to apply gas on its own (latest master code, so without pedal)

rinterdonato, 2025-01-20:
> maybe your car only allowed enabling at 22 mph:

Shane,

Sorry for the delay in responding. I have made the suggested change and it
has no effect. However, I have tried some other things, and it turns out
that the latest version of  SunnyPilot 0.9.71 restores the functionality
for my 2018 Acura RDX. Not sure why, but it allows me to engage while the
car is not moving when there is a lead car or RED stoplight. Also, it
allows me to reduce the cruise control speed down to 1 mph (although the
engage speed shown on the UI only goes down to 25mph).

I will try your suggestion again when 0.9.8 comes out.

Thanks for your help,

Rich

closed 2025-09-07 by sshane