commaai/opendbc #1630: 2021 KIA CEED Car Unrecognized

justin927, 2025-01-13:
### Describe the bug

I connected the Comma 3x with a 2021 KIA CEED but got the "car unrecognized" message, even after trying both the release3 and nightly versions, which resulted in the same issue.

### Provide a route where the issue occurs

5e45e04bc1594710/00000000--99cd0e750c

### openpilot version

0.9.8/mater-ci

sshane, 2025-01-13:
Add `export FINGERPRINT=KIA_CEED` to the top of `launch_openpilot.sh` and report back with a route please.

justin927, 2025-01-14:
> Add `export FINGERPRINT=KIA_CEED` to the top of `launch_openpilot.sh` and report back with a route please.

Thanks for your suggestion! I added export FINGERPRINT=KIA_CEED to the top of launch_openpilot.sh as you recommended, and it worked perfectly. Here's the route ID for your reference: 5e45e04bc1594710/00000005--eb45799ba2.

Thanks again!

sshane, 2025-01-15:
https://github.com/commaai/opendbc/commit/5d30ef516889295c61bee67cf5ed51ef76e076aa

closed 2025-01-15 by sshane