commaai/opendbc #1637: Honda: Third-gen facelift Acura MDX needs a separate car port

jyoung8607, 2025-01-14:
The [Acura MDX got a facelift for model-year 2022](https://en.wikipedia.org/wiki/Acura_RDX#2021_facelift), which came with a new camera/radar architecture. This architecture is also shared by newer Honda Odyssey.

Correctly supporting these cars requires a new model port with minor platform changes. The CAN messaging is nearly identical, but there are architectural changes that affect the minimum speed for lateral control and cluster display of the cruise setpoint. A port was started in commaai/openpilot#25481 by @csouers but wasn't quite completed or merged.

In the meantime, starting with commaai/openpilot#27812, we inadvertently started recognizing them as pre-facelift cars. So we need the MY2022 port **and** then we need to disentangle all the firmware that was added to ACURA_RDX_3G. This will be more than just reverting the one PR, the fuzzy fingerprinting bot may have added more on top.

- [ ] Temporarily revoke documented support for MY2022 Acura MDX
- [ ] Merge a working port for MY2022 and forward
- [ ] Disentangle UDS firmware between Acura MDX 3G and 3G-facelift
- [ ] Restore MY2022 and forward Acura MDX as officially supported

mvl-boston, 2025-01-14:
Thank you, for merging.   Let me know if you would like me to propose code for this.   In the past week I have looked through all the items mentioned and can follow everything except the fuzzy logic, and can support as needed.

There is also a problem with the longitudinal acceleration/develeration tuning parameters and I do not know if that is unique to the 2022 RDX or also impacts other bosch models.

mvl-boston, 2025-01-16:
Issue 1638 links to the RDX information.

The MDX is a separate Acura vehicle that has its own separate incompatibility issues.    The subject line for this issue and linked issue #1638 should say RDX.

jyoung8607, 2025-01-16:
My mistake, I'll sort it out shortly. Thanks!

mvl-boston, 2025-02-19:
Port is about 75% complete:
- stable (but not properly fingerprinted 0.97 openpilot) code - https://github.com/mvl-boston/openpilot/tree/RDX_stable_097
- 0.98 opendbc port in process - https://github.com/mvl-boston/opendbc

I don't have edit rights to this bug but it still improperly says RDX.

Remaining tasks:
- Learn how to use the simulator to pass relevant safety testing (and what that testing is)
- Testdrive the opendbc version
- Look into the repeated missing vin in the error log (might need a new DBC, will poke around cabana)

mvl-boston, 2025-02-20:
Confirmed fingerprinting bot didn't add any new firmwares.

And got my.code to compile on the 3x.   Hope to testdrive by tomorrow.

Anything else I meed to do before a pr?

mvl-boston, 2025-02-22:
Looks like I'm crashing the canbus due to poorly performing code (CPU maxed out) on my first attempt, and my attempt of mirroring the approaches of https://github.com/commaai/openpilot/pull/25481 are either no longer fingerprinting or crashing before fingerprinting occurs.

At this point I'm blocked on further debugging due to unavailability of logs because of https://github.com/commaai/openpilot/issues/34666

If anyone wants to finish off the fix, my work to date is at:
https://github.com/mvl-boston/opendbc

mvl-boston, 2025-03-04:
I have been making substantial progress on this with the help of Jeff Liu (an Odyssey driver).   I found a canbus signals that permits auto-detection of both the speed limited cruise warnings, plus autodetection of metric vs imperial cruise speed, and both are in an 0.98 port nearing PR status.   We're testing to see if these cars have steer maximums before finishing the pr's (and will move to 0.99 based code).

Prior cpu max issues were just a symptom of some syntax errors, not poor performance.   Seems like 0.97 likes to endless retry code syntax problems uselessly killing CPU.   Thankfully 34666  is now fixed.

mvl-boston, 2025-07-27:
Just a reminder, that title and narrative should be changed from MDX to RDX.