commaai/opendbc #1718: Tesla HW4 improvements

sshane, 2025-02-13:
- [ ] There are faults we don't handle with FSD long and openpilot lateral. We can 1) Detect when FSD is turned on and warn user to switch to TACC, or 2) Automatically send a message to switch to TACC (Hyundai CAN FD does something similar with the ADAS ECU).
- [ ] Temporary steering fault when overriding lateral very forcefully. Does the fault signal go high with FSD? If so, ignore in this case, else figure out how to not fault the steering.

adeebshihadeh, 2025-02-15:
-  [ ] FSD still does the blinkers (and who knows what else)

lukasloetkolben, 2025-02-22:
- [ ] Passthrough ASS / Autopark / Lane Keeping

sshane, 2025-03-11:
Summon goes into the autopark/summon state before ACC enables, but Autopark enables ACC right away so we currently can't know to forward it. Maybe we need to allow some delay on our cancel command for a few frames so we have some time to read the autopark state.

closed 2025-05-19 by sshane