commaai/opendbc #1724: Rivian port issues + improvements

adeebshihadeh, 2025-02-16:
We'll have a Rivian for a few days next week at [COMMA_HACK](https://github.com/commaai/opendbc/issues/1625).

To make the most of that time, post any quirks, bugs, issues, and feedback here. Ideally, include at least your Dongle ID and preferably a route too.

cw-cw-cw, 2025-02-16:
Driver+ engagement seems to be disabled when OpenPilot is available. Double click on the stalk didn’t turn on Driver+ (the icon showed available). However, when my Openpilot was unavailable (had to remount my unit because I wasn’t patient enough to wait 48 to let the adhesive cure, so it needed to recalibrate) then Driver+ turned on right away (and ironically the unit recalibrated while Driver+ was running). 

Not sure what ideal behavior should be - just making a note of my experience.

adeebshihadeh, 2025-02-17:
Looks like Driver+ is just the name for their whole ADAS suite - is there something specially you're missing? Generally, openpilot replaces the built-in lane centering and adaptive cruise.

cw-cw-cw, 2025-02-17:
Good clarification - sorry about that - when I mentioned Driver+ I meant the “Highway Assist” - which is Rivian’s full ADAS on mapped roads.  Typically, pull down once on the stalk and it activates ACC (no LKA), and double pull on the stalk turns on “Highway Assist” (ACC+LKA on mapped roads). 

FWIW - I have no other experience with Openpilot prior to this, so forgive my ignorance.  I just wanted to contribute the community with my experiences.

closed 2025-02-25 by sshane

sshane, 2025-03-06:
@cw-cw-cw do you have this route where Driver+ was working while openpilot was calibrating?