commaai/opendbc #1754: Q4 fingerprint

blue-genie, 2025-02-20:
Not yet officially supported, just getting ready for Hackaton:

It seems that sporadically errors similar to the below one are logged:
```
[2.972794] isotp_parallel_query.py:146 - : iso-tp query bad response: (1990, None) - 0x7f2231
[2.972962] isotp_parallel_query.py:146 - : iso-tp query bad response: (1990, None) - 0x7f2231
[2.973382] isotp_parallel_query.py:146 - : iso-tp query bad response: (1840, None) - 0x7f2231
[2.973458] isotp_parallel_query.py:146 - : iso-tp query bad response: (1840, None) - 0x7f2231
[2.973865] isotp_parallel_query.py:146 - : iso-tp query bad response: (1988, None) - 0x7f2231
[2.973961] isotp_parallel_query.py:146 - : iso-tp query bad response: (1988, None) - 0x7f2231
```

Which results in no fingerprinting.

Route: e36b272d5679115f/00000307--2e7636e703

`./selfdrive/debug/fingerprint_from_route.py e36b272d5679115f/00000307--2e7636e703/0` returns

```
found 33 messages. CAN fingerprint:

65: 8, 66: 8, 73: 8, 119: 8, 125: 8, 126: 8, 130: 8, 145: 8, 146: 8, 357: 8, 359: 8, 390: 8, 391: 8, 512: 8, 514: 8, 516: 8, 517: 8, 531: 8, 532: 8, 534: 8, 535: 8, 560: 8, 578: 8, 936: 8, 937: 8, 943: 8, 972: 8, 976: 8, 981: 8, 1045: 8, 1070: 8, 1111: 8, 1200: 8

FW fingerprint:

Traceback (most recent call last):
  File "/DevWork/op/./selfdrive/debug/fingerprint_from_route.py", line 46, in <module>
    get_fingerprint(lr)
  File "/DevWork/op/./selfdrive/debug/fingerprint_from_route.py", line 32, in get_fingerprint
    for f in fw:
TypeError: 'NoneType' object is not iterable
```

kill openpilot and restart results in successful fingerprinting, route: `e36b272d5679115f/00000308--b61d26edc1/0`

right before the bad fingerprinting I updated `cereal/services.py`, started `launch_openpulot.py` and I noticed that scons had work to do. After bad fingerprinting i killed `launch_openpilot.py` and reran it, scons had no work to do and it fingerprinted correctly.

sshane, 2025-02-20:
That route is only a couple seconds long, so it could still have fingerprinted if you left it. This route is a better example: e36b272d5679115f/00000302--1a2a40e714

sshane, 2025-02-20:
Please do not report issues with dirty git diffs/local changes that you don't understand. card crashed due to local changes related to the radar interface

sshane, 2025-02-20:
https://github.com/commaai/openpilot/pull/34579 should improve this by showing an alert that something crashed

closed 2025-02-20 by sshane

blue-genie, 2025-02-20:
> Please do not report issues with dirty git diffs/local changes for in-progress work. card crashed due to local changes related to the radar interface
> 
> ```
> Traceback (most recent call last):
>   File "/data/openpilot/openpilot/system/manager/process.py", line 41, in launcher
>     mod.main()
>   File "/data/openpilot/selfdrive/car/card.py", line 286, in main
>     car = Car()
>           ^^^^^
>   File "/data/openpilot/selfdrive/car/card.py", line 102, in __init__
>     self.RI = get_radar_interface(self.CI.CP)
>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/data/openpilot/opendbc/car/car_helpers.py", line 160, in get_radar_interface
>     return RadarInterface(CP)
>            ^^^^^^^^^^^^^^^^^^
>   File "/data/openpilot/opendbc/car/ford/radar_interface.py", line 128, in __init__
>     elif self.radar == RADAR.STEER_ASSIST_DATA:
>                        ^^^^^^^^^^^^^^^^^^^^^^^
> AttributeError: type object 'RADAR' has no attribute 'STEER_ASSIST_DATA'
> ```

Yes, I'm aware that other routes aren failing because local changes, and his is why I only mentioned routes 00000307 and 00000308. 

I was not aware that once the dash-cam message for unrecognized car is displayed, it can still be fingerprinted.

sshane, 2025-02-20:
Those routes had a exception as well

blue-genie, 2025-02-20:
On `e36b272d5679115f/00000308--b61d26edc1` I drove for 10 minutes with OP in control, The only difference between 307 and 308 was a reboot.

When it fails to fingerprint I will let it run longer.