commaai/opendbc #1794: Ford Q4 fingerprinting issue - take 2

blue-genie, 2025-02-22:
Route: e36b272d5679115f/0000030f--340e67d898

More than 6 minutes long - Dashcam with unrecognized car

`git diff` output:

```
comma@comma-e11dd515:/data/openpilot$ git diff
diff --git a/opendbc_repo b/opendbc_repo
--- a/opendbc_repo
+++ b/opendbc_repo
@@ -1 +1 @@
-Subproject commit 4185f81635edab3e943a881edde7e1a03d43ec9c
+Subproject commit 4185f81635edab3e943a881edde7e1a03d43ec9c-dirty
```

```
comma@comma-e11dd515:/data/openpilot/opendbc_repo$ git diff
diff --git a/opendbc/car/ford/interface.py b/opendbc/car/ford/interface.py
index 26a18ea..1b22341 100644
--- a/opendbc/car/ford/interface.py
+++ b/opendbc/car/ford/interface.py
@@ -20,7 +20,7 @@ class CarInterface(CarInterfaceBase):
   @staticmethod
   def _get_params(ret: structs.CarParams, candidate, fingerprint, car_fw, experimental_long, docs) -> structs.CarParams:
     ret.brand = "ford"
-    ret.dashcamOnly = bool(ret.flags & FordFlags.CANFD)
+#    ret.dashcamOnly = bool(ret.flags & FordFlags.CANFD)

     ret.radarUnavailable = Bus.radar not in DBC[candidate]
     ret.steerControlType = structs.CarParams.SteerControlType.angle
```

To fix it, while driving I just rebooted the device. The next route `e36b272d5679115f/00000310--6760d21c98` did a good fingerprinting.

adeebshihadeh, 2025-02-23:
I don't understand this issue?

sshane, 2025-02-23:
Try this branch for a few days: https://github.com/commaai/opendbc/tree/ford-queries

The query for the EPS didn't come back the second time, unclear why the panda or gateway is dropping messages silently (we see them on the other buses, but not the one we requested it on).

blue-genie, 2025-02-23:
route: e36b272d5679115f/00000319--0e9ee82ed2 - using the `ford-queries` branch. 

Rebooted, and route 31a worked fine. 

```
/data/openpilot
++ git diff
diff --git a/opendbc_repo b/opendbc_repo
index 4ffe21a51..0ebf0fbe4 160000
--- a/opendbc_repo
+++ b/opendbc_repo
@@ -1 +1 @@
-Subproject commit 4ffe21a516c38862da6c827b98fb2f242d1f66de
+Subproject commit 0ebf0fbe42c7ea1494d1a821275b9e20732310fc
++ git rev-parse HEAD
088ba13f214ac7b44b9b1f6daf6e159d649dfd88
```