commaai/opendbc #1834: Move CAN ignition hooks and add test

adeebshihadeh, 2025-02-24:
CAN ignition lives in [board/drivers/can_common.h](https://github.com/commaai/panda/blob/master/board/drivers/can_common.h). This is the last car-specific code in panda and should be moved here.

It should also be possible to fully validate these signals in CI. Assert that the hook doesn't unintentionally trip on all the other cars, etc.

sshane, 2025-02-24:
with proper rx checks (counter, checksum, etc.) to reduce chances of overlap between cars

adeebshihadeh, 2025-02-24:
> with proper rx checks (counter, checksum, etc.) to reduce chances of overlap between cars

Let's do this as a separate issue once this is in.

sshane, 2025-03-03:
@ccdunder want to tackle this one?

aubsw, 2025-04-15:
@sshane I humbly propose https://github.com/commaai/opendbc/pull/2131

Lemme know what you think!

aubsw, 2025-04-26:
I've been [experimenting](https://github.com/commaai/openpilot/pull/35082) with adding coverage to test_models.py

We actually have some decisions we need to make here:

My first thought was to check every rx from every log to ensure that only [gm|mazda|rivian|tesla] see CAN-ignition. i.e.:

```diff
diff --git a/selfdrive/car/tests/test_models.py b/selfdrive/car/tests/test_models.py
index 53f2bb586..48b33b42b 100644
--- a/selfdrive/car/tests/test_models.py
+++ b/selfdrive/car/tests/test_models.py
@@ -243,6 +243,11 @@ class TestCarModelBase(unittest.TestCase):
         if self.safety.safety_rx_hook(to_send) != 1:
           failed_addrs[hex(msg.address)] += 1

+        if self.CP.brand in ("gm", "mazda", "rivian", "tesla"):
+          self.assertTrue(self.safety.get_ignition_can())
+        else:
+          self.assertFalse(self.safety.get_ignition_can())
+
       # ensure all msgs defined in the addr checks are valid
       self.safety.safety_tick_current_safety_config()
       if t > 1e6:
```

https://github.com/commaai/openpilot/blob/c9c6eaa725ed9f2a3c0dd98d844979ccdf358fea/selfdrive/car/tests/test_models.py#L237-L249

But there are two problems with this:

1. `assertFalse` on every rx blows up the test run time from 6 mins to 15 mins
2. Not every single message from the [gm|mazda|rivian|tesla] logs have ignition_can == True 
   - see [CI failure](https://github.com/commaai/openpilot/actions/runs/14683716070/job/41209614451?pr=35082))

@sshane do you have a preferred way to deal with each of these problems? Here are some options:

For problem 1:
- (a) sample the messages in a deterministic way (to reduce total number of asserts)
- (b) instrument opendbc with a counter or a latch that triggers whenever CAN-ignition is detected, check this after all rx's instead of after each

For problem 2:
- (a) find different logs for [gm|mazda|rivian|tesla] where every rx has CAN-ignition 
- (b) same as option (b) above

lmee, 2026-02-15:
I reviewed "Move CAN ignition hooks and add test" and read all comments before proposing a patch path.

Problem I see: CAN ignition lives in . This is the last car-specific code in panda and should be moved here.
Root-cause hypothesis: a state flow mismatch between action handling and final UI state.
Useful clue from comments: I've been with adding CAN ignition coverage to test models.py In the experiment, we check every rx to ensure that only [gm|mazda|rivian|tesla] see CAN ignition. I.e.: diff diff --g.
First patch plan: reproduce the issue against acceptance criteria, isolate the root cause in the relevant module, and submit a focused fix with regression tests.
If this direction matches what you want, I can publish repro notes first so review starts from concrete evidence.

genesis-ai-labs-star, 2026-03-10:
Starting on this. Quick plan: reproduce locally, draft minimal patch + tests, then open PR for review. Please confirm any bounty/acceptance criteria if needed.

zscy0710, 2026-03-11:
Quick update: implementation is complete and PR is up at #3185: https://github.com/commaai/opendbc/pull/3185 (with ignition hook migration + CI tests, currently green). 

For bounty payout method/timeline, maintainer confirmation is still needed on this issue.

seraphim941, 2026-03-11:
I'd like to work on this bounty. I'll move the CAN ignition hooks and add a test as specified.

genesis-ai-labs-star, 2026-03-14:
⭐ Star on it. Pulling repo to repro CAN ignition hooks issue. Will propose PR with tests.